trezor/trezor-firmware #824: QR code button is all white in 1.9.0

bosomt, 2020-02-04:
**Describe the bug**
QR code button is all white in 1.9.0 it should be QR code

**Firmware version or revision**
1.9.0

**Desktop/smartphone setup (please complete the following information):**
Suite and beta wallet
latest chrome

**To Reproduce**
Steps to reproduce the behavior:
go to wallet and hit show full address

**Expected behavior**
Button must say  QR code

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
[image]

prusnak, 2020-02-04:
Fixed in 0ceba808760a61b014f1a0bee9a22919ffdee6d4

closed 2020-02-04 by prusnak

bosomt, 2020-02-05:
works fine in latest master build

bosomt, 2020-07-20:
**QA OK**

FW 1.9.1 0xc6b2580cd245ee924507f45e9675f857a3d78768,
wallet b3f18eb73765f92dd35eff2c562ef1edb5dd9fa3