trezor/trezor-firmware #825: Adding the support of Credits (CS) blockchain platform

ArsenyFirst, 2020-02-04:
Website - https://credits.com/
Github - https://github.com/CREDITSCOM

1) Our app can be compared to [Tezos app ](https://github.com/trezor/trezor-firmware/tree/master/core/src/apps/tezos) in terms of scale. We also use the same elliptic curve - ed25519
2) No, you don't
3) Yes. We have Web Wallet with support of our Mainnet. Here is the [link](https://wallet.credits.com/).
4) Address format - ed25519 + base58. The signing of the transaction - ed25519
5) [Credits Blockchain Explorer](https://monitor.credits.com/)

prusnak, 2020-02-04:
We are not adding coins unless they are in the TOP 30 on Coinmarketcap.

closed 2020-02-04 by prusnak

ArsenyFirst, 2020-02-04:
Hello! Thanks for your fast reply. 

As we saw you have added the Wanchain support recently. It is not in the top 30 CMC as well. Can you please clarify this situation?

matejcik, 2020-02-04:
wanchain support was added more than two years ago, before this policy was in place