trezor/trezor-firmware #826: trezor-core event loop survives SystemExit

matejcik, 2020-02-04:
[here](https://github.com/trezor/trezor-firmware/blob/master/core/src/trezor/wire/__init__.py#L412) we claim that "handle_session should never exit". I'm not convinced that's the case; notably, SystemException gets silently eaten, which seems wrong.

(in contrast, `_step()` in loop.py only catches Exception subclasses, but BaseException will kill it.)

I think that the line in question should not be catching BaseException, just Exception. Moreover, I think it should be possible to do a controlled shutdown of `handle_session` (and there should probably be a special exception for that purpose, but not necessarily derived from BaseException).

The other exceptions that are derived from `BaseException` but not `Exception` are `KeyboardInterrupt` (we don't care) and `GeneratorExit`. The latter is an internal signal and should not propagate, but we do use it explicitly in places, so who knows.

related to #727 where this issue allows the emulator to survive a `error_shutdown()`

matejcik, 2020-02-04:
related: should `error_shutdown()` on the emulator boil down to `sys.exit()`? I don't think so, but perhaps there is a good reason?
(see: https://github.com/trezor/trezor-firmware/blob/75264a07a883dc1c6494442b9e3bf28f08fc021e/core/embed/unix/main.c#L417-L422)

closed 2020-02-13 by matejcik

tsusanka, 2020-02-17:
@matejcik could you write some test scenarios or mark this one as `doesn't need QA`?

matejcik, 2020-02-17:
marking "doesn't need QA", but noting that this is a system-wide effect, so all the existing QA scenarios are technically testing that the modification works fine

not sure if we have "cancellation" qa scenarios though? something like, "run an operation, try to interrupt it on the host in the middle", reload wallet, click cancel, etc?

tsusanka, 2020-02-17:
I don't think we have, but it's a good idea. I'll forward it to QA.

sorooris, 2020-02-17:
QA implicitly tests negative scenarios, including cancelling modals.

PR #840 (2020-02-13): wipe code activation tests
this adds wipe code activation test for T1 and TT (fixes #727)

in relation, the emulator restart issue is resolved (fixes #826)

i have also simplified the wipe code setup test for T1 with a new debuglink feature