trezor/trezor-firmware #829: SD protect: Allow PIN entry even when card is not present

andrewkozlik, 2020-02-05:
When SD protect is enabled but the SD card is not present, Trezor requests SD card insertion before prompting for PIN. This is a problem if the user wants to enter their wipe code to quickly wipe the device and doesn't have enough time to find the SD card or switch to bootloader mode.

Trezor should:
1. Prompt for PIN.
2. If the SD card is not present, then check for wipe code and request SD card.
3. Attempt to unlock the device.

closed 2020-03-19 by andrewkozlik

andrewkozlik, 2020-03-19:
Test cases for @trezor/qa :

TC1
- Enable PIN, SD protect and wipe code.
- Remove the SD card and invoke a function that requires PIN entry (try unlock, change PIN, dry run recovery and fido2 login).
- Ensure that you are able to enter the wipe code without the presence of the SD card in all cases.

TC2:
- Enable PIN, SD protect and wipe code.
- Ensure that unlock, change PIN, dry run recovery and fido2 login function correctly when the SD card is present.

TC3:
Ensure that unlock works correctly for various combinations of the three features (PIN / SD protect / wipe code) enabled and disabled.

andrewkozlik, 2020-03-19:
@trezor/qa I recommend to hold off on testing this until https://github.com/trezor/trezor-firmware/pull/893 is merged, so that you don't have to test it separately.

bosomt, 2020-03-20:
@andrewkozlik waiting for test sheet

bosomt, 2020-03-23:
@andrewkozlik works fine ! of course TC3 will take more time and we will try to break it during this week testing

sorooris, 2020-03-26:
TC1 and TC2 tested, still on TC3.

sorooris, 2020-03-30:
Tested combinations:
PIN + SDP
PIN + WC
PIN + SDP + WC
SD + WC
SD
WC
- on revision **8ae0535e** and is **OK**.