trezor/trezor-firmware #839: Passphrase: Cache last X sessions

tsusanka, 2020-02-10:
Now with the new passphrase redesign we can easily cache more passphrases at once, indexed by the session id.

We will replace `_cache_session_id` and `_cache` in `storage.cache` with a map indexed by the session ids:

```
_caches = {
  "[session id A]": { ... },
  "[session id B]": { ... },
  "[session id C]": { ... }.
  "[session id D]": { ... },
}
```

On `Initialize(session_id=None)` we should clear the least used session id. We will not provide a way to clear all sessions.

How many should we store?

prusnak, 2020-02-10:
What is the size of the data stored in the cache (per item)? Depending on that I'd go for 8 or 16 entries.

Implementing LRU in Python is pretty straightforward (just remove the currently used value from cache and put it back to the top, on cache exhaustion remove the last item). For C I would recommend a counter which increases with every cache use and storing the current value to the item currently being used. On cache exhaustion, we'd remove the item with the lowest stored counter value.

matejcik, 2020-02-11:
the data currently used is: seed, seed without passphrase, Cardano root (if in use) and a flag for Monero live refresh (if in use)

so in a normal scenario with BIP39, that's 32 bytes session id + 2*64 bytes for seed data

prusnak, 2020-02-11:
So 160 bytes per entry. Maybe using 10 entries is OK? That is ~1600 bytes of RAM.

tsusanka, 2020-02-14:
Note to myself: let's investigate the "lower level" caches (in trezor crypto).

tsusanka, 2020-02-18:
I have removed the BIP-39 and it seems to me there is no performance difference.

tsusanka, 2020-02-21:
QA scenarios:

Wait till "new passphrase" is implemented in Wallet. Then test against both Wallet and Suite.

For both TT and T1:
- Start a few hidden wallets one after another. Switch to the first hidden wallet, the passphrase should not be prompted.
- Create 10 hidden wallets. Create one more. Go to the first one. The passphrase should be prompted. Go to the third one. Passphrase should not be prompted.

bosomt, 2020-03-02:
Works fine with 2.3.0 and latest web dev suite

bosomt, 2020-03-02:
doesnt work with 1.9.0 it asks for passphrase even there are only 3 passphrase wallets

tsusanka, 2020-03-02:
I can confirm - seems to work with 2.3.0 but asks for passphrase repeatedly in 1.9.0. It seems to me I am receiving `Initialize.session_id = null` after changing to another hidden wallet. @szymonlesisz could you look into this?

tsusanka, 2020-03-03:
I created https://github.com/trezor/trezor-suite/issues/1272 to tackle this in Suite.

tsusanka, 2020-03-20:
Turns out this is not bug. See https://github.com/trezor/trezor-suite/issues/1272#issuecomment-601605764 for explanation.

closed 2020-03-20 by tsusanka