trezor/trezor-firmware #841: The final answer to the USB Chrome pop up dialog

tsusanka, 2020-02-11:
This was brought up by @matejzak and the Suite team again.

[image]

------

We went back and forth on the Chrome pop up dialog. Removing it at some point and adding it back later because of phishing concerns if I remember correctly.

Many users find this annoying. Also with the introduction of Suite it does not make any sense anymore, does it?

I suggest to remove the pop up completely or we could show it only if the device is new (= no firmware installed).

prusnak, 2020-02-11:
The dialog is only shown when there is a Chrome browser running and honestly, I think this should be a setting in Chrome whether to show these.

That said, showing the pop-up only if the device is new (only bootloader with no firmware) is acceptable from where I stand.

tsusanka, 2020-02-11:
> The dialog is only shown when there is a Chrome browser running and honestly, I think this should be a setting in Chrome whether to show these.

Yet that is true, how many users do not have a Chrome browser running in the background these days..

matejzak, 2020-02-11:
> we could show it only if the device is new (= no firmware installed) to simplify the onboarding and downloading of Suite.

Sounds like a perfect solution to me.

prusnak, 2020-02-12:
PR in #845

PR #845 (2020-02-19): Don't show WebUSB popup in firmware, in bootloader only if no firmware present
Fixes #841
closed 2020-02-19 by prusnak