trezor/trezor-firmware #842: Document required changes after passphrase redesign

prusnak, 2020-02-11:
After closing #1 we should document what are the necessary changes in the API a wallet integrating Trezor needs to change. I imagine having a document in docs which will describe migration path from supporting T1/TT before and after the change.

tsusanka, 2020-02-11:
I wrote [passphrase.md](https://github.com/trezor/trezor-firmware/blob/master/docs/common/communication/passphrase.md) and [sessions.md](https://github.com/trezor/trezor-firmware/blob/master/docs/common/communication/sessions.md). But it is true it does not explicitly explain a migration.

prusnak, 2020-02-12:
Motivation: I want to have a go-to document for people implementing 3rd party Trezor support (such as Blockstream Green, Monero, WallETH, etc.) where we can send them. We'll handle the Suite/Connect/Wallet/Electrum updates by ourselves.

matejcik, 2020-02-18:
Resolved via 02f2158

closed 2020-02-18 by matejcik