trezor/trezor-firmware #843: UI testing does not deal with testcase removal

matejcik, 2020-02-12:
When a testcase is removed, the matching `fixtures.json` key is not removed, and `pytest --ui=test` runs without error.

In CI this leads to a later failure in `prepare_ui_artifacts.py`, which cannot find the corresponding `recorded/` directory for the missing test case.

tsusanka, 2020-02-12:
A hotfix is to run `--ui=record` as you probably figured out. I guess we should indeed check if all hashes were "used" during `ui=test`.

matejcik, 2020-02-12:
it's not, as it turns out! `--ui=record` does not seem to remove missing fields

tsusanka, 2020-02-12:
Hmm, I think I broke it when we made the JSON transition. I'll fix it!

tsusanka, 2020-02-17:
@matejcik I am wondering how this should work if you select a subset of tests (i.e. `pytest -s tests/device_tests/some_particular_test.py --ui=test` or `-k` usage).

In that case there are items in fixtures.json that were not tested obviously - by design. Should we do this check only if the target is the full folder? Or provide a new pytest option `--ui-check-missing`? The second sounds like a more robust solution.

matejcik, 2020-02-17:
Yes, a separate option seems like a better design. Raise in case of ui=test, remove missing cases when ui=record
we could want some sort of confirmation if there are more than 10 testcases removed, but I suppose we have git for that :)

PR #857 (2020-02-18): Introduce --ui-check-missing to test/remove missing tests
Closes #843
closed 2020-02-18 by matejcik