trezor/trezor-firmware #854: 94e44a0 breaks the emulator on macOS

leshik, 2020-02-15:
This commit https://github.com/trezor/trezor-firmware/commit/94e44a08e1495b3e9c299302446d0b56b58d4db2 breaks the emulator on `macOS` (which lacks `inotify`):
```
$ ./emu.py
Traceback (most recent call last):
  File "./emu.py", line 20, in <module>
    import inotify.adapters
  File "/Users/leshik/.local/share/virtualenvs/trezor-firmware-w0YaNlwH/lib/python3.7/site-packages/inotify/adapters.py", line 11, in <module>
    import inotify.calls
  File "/Users/leshik/.local/share/virtualenvs/trezor-firmware-w0YaNlwH/lib/python3.7/site-packages/inotify/calls.py", line 39, in <module>
    inotify_init = _LIB.inotify_init
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ctypes/__init__.py", line 377, in __getattr__
    func = self.__getitem__(name)
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ctypes/__init__.py", line 382, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: dlsym(0x117c55840, inotify_init): symbol not found
```

prusnak, 2020-02-15:
@matejcik maybe we can replace inotify with watchdog? https://pypi.org/project/watchdog/

leshik, 2020-02-15:
@prusnak Maybe it's worth to use [fswatch](https://github.com/emcrisostomo/fswatch) abstraction layer?

prusnak, 2020-02-15:
watchdog does that already

matejcik, 2020-02-17:
Fixed by catching `Exception` instead of just `ImportError` when inotify fails. inotify was already optional, only required for the `-w` option that restarts the emulator -- which didn't work on OSX before either ;)

The drawback of watchdog is that it introduces threading, which i'm honestly not sure what would do in a script that already tries hard enough to pretend to be a shellscript. I'd accept a PR adding it, but it doesn't seem worth the effort for now.

closed 2020-02-17 by matejcik

prusnak, 2020-02-17:
ACK

commit 1d41141a (2020-02-17): core/emulator: properly ignore inotify problems (fixes #854)