trezor/trezor-firmware #855: Match the XPUBs on the device with the ones on the client

prusnak, 2020-02-16:
**Describe the bug**

Reported here: https://twitter.com/_benma_/status/1227900614505566208
and here: https://twitter.com/_benma_/status/1227912076368252928

There are two issues with GetAddress for multisig:

- [ ] when Electrum uses Zpub, Trezor shows xpub - we should be showing the same value
- [ ] we should indicate which of the shown XPUBs belongs to the device

A clear and concise description of what the bug is.

**Firmware version or revision**
Describe firmware version or revision you are using when encountering the bug.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [e.g. Trezor Wallet, Trezor Beta Wallet, Trezor Suite, Electrum]
 - OS: [e.g. Windows, macOS, Linux, Android]
 - Browser [e.g. chrome, firefox, safari]
 - Version [e.g. 42]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

matejcik, 2020-02-17:
re (1) what Electrum shows as "master public key" seems unrelated to the XPUB in the address. Although the `show_xpubs` method does not reflect the address type correctly (legacy / p2sh / segwit), the correct result will be different from the "master public key" anyway

prusnak, 2020-02-18:
2) is fixed for TT in https://github.com/trezor/trezor-firmware/pull/860

prusnak, 2020-02-19:
The second issue is fixed for T1 in #862

prusnak, 2020-02-19:
The first issue is impossible to be fixed without overhauling the multisig protocol. Putting back to backlog.

prusnak, 2021-01-13:
First issue has fix in PR https://github.com/trezor/trezor-firmware/pull/1415

closed 2021-01-21 by prusnak