trezor/trezor-firmware #866: Encoding error on Stellar change_trust operation

michailbrynard, 2020-02-20:
**Describe the bug**
When signing a stellar transaction that includes a change trust operation, trezorctl throws an error: `AttributeError: 'bytes' object has no attribute 'encode'`

**Firmware version or revision**
trezorctl, version 0.11.6

**Desktop/smartphone setup (please complete the following information):**
 - OS: macOS

**To Reproduce**
`trezorctl stellar-sign-transaction AAAAAGXJo93R2kgzDMosAQYUzs3+D8MnCvBD98+m8rpQ4bdQAAAAZAGwANcAAAABAAAAAAAAAAAAAAABAAAAAAAAAAYAAAABTEZFQwAAAAAN4stij7kEnWdNPnbegeKCu9RlgEd2dWbWrOZ8QNImkH//////////AAAAAAAAAAA=`

Produces:
```
Please confirm action on your Trezor device
Traceback (most recent call last):
  File "/usr/local/miniconda3/envs/general/bin/trezorctl", line 11, in <module>
    sys.exit(cli())
  File "/Users/michail/.local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/michail/.local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/michail/.local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/michail/.local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/michail/.local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/michail/.local/lib/python3.7/site-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/cli/stellar.py", line 76, in sign_transaction
    resp = stellar.sign_tx(client, tx, operations, address_n, network_passphrase)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/stellar.py", line 368, in sign_tx
    resp = client.call(operations.pop(0))
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/tools.py", line 233, in wrapped_f
    return f(client, *args, **kwargs)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/client.py", line 211, in call
    resp = self.call_raw(msg)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/client.py", line 144, in call_raw
    self._raw_write(msg)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/client.py", line 149, in _raw_write
    self.transport.write(msg)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/transport/bridge.py", line 161, in write
    protobuf.dump_message(buffer, msg)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/protobuf.py", line 427, in dump_message
    dump_message(counter, svalue)
  File "/usr/local/miniconda3/envs/general/lib/python3.7/site-packages/trezorlib/protobuf.py", line 421, in dump_message
    svalue_bytes = svalue.encode()
AttributeError: 'bytes' object has no attribute 'encode'
```

michailbrynard, 2020-02-20:
I've submitted a fix here https://github.com/trezor/trezor-firmware/pull/865

tsusanka, 2020-02-21:
Note: we would catch this in https://github.com/trezor/trezor-firmware/issues/2

matejcik, 2020-02-21:
> Note: we would probably catch this in #2 :/

that is unlikely, actually; AFAICT the bug is in the part of code that decodes the transaction from a binary representation *before* it is sent to Trezor. Even the existing tests skip the decoding step in favor of sending pre-parsed transaction data.

matejcik, 2020-03-02:
fixed in c2635a47732cc5426e311603b2c15452fe5fafdb

closed 2020-03-02 by matejcik