trezor/trezor-firmware #867: Send

tsusanka, 2020-02-21:
We have discussed this already, but since we try to "make Suite people happy" and @szymonlesisz brought it up again, I would like to have a clear and tracked decision on this.

Connect now sends GetAddress twice. Once without the display parameter to obtain the address and second time to display it on Trezor. This is required so the user can check the address on Trezor and in Connect.

Options:
- Do nothing. We do not want to modify the protocol in any way and calling the message twice seems fine (although it slows Connect a bit).
- Somewhat modify Button Requests to allow including a message. We have discussed this a few times and it seems like a way bigger issue.
- Other ideas?

In a nutshell, it sounds like we do not want to do this, but I would like to hear if there are other ideas. @prusnak @matejcik

matejcik, 2020-02-21:
what is the usecase exactly?

prusnak, 2020-02-21:
If you call `GetAddress` with `show=True` you don't know the address yet. You learn it only after user confirms that.

So what wallets do is they call `GetAddress(show=False)` to obtain the address, then `GetAddress(show=True)` to confirm it. This is obviously not optimal, but can't be fixed easily without introducing arbitrary data fields to ButtonRequest.

matejcik, 2020-02-21:
transmitting this information through a `ButtonRequest` seems like a wrong idea in any case; given that this would be "arbitrary data", host cannot rely on the address being there. (imagine that for some reason a different screen pops up that doesn't contain the address, or contains a different one). Even if the ButtonRequest data has semantics, it's still basically "looking at the screen" as opposed to doing a protocol.

A fully proper fix would be something like `GetAddress` -> `AddressRequest` -> `AddressAck` -> (screen is shown here) -> `Success`
Which ... seems very similar to `GetAddress -> Address`, plus `GetAddress(show_display=true)` -> (screen is shown here) -> `Address`

alternately, `GetAddress` could be used to send the address to host, and `ShowAddress` for displaying it for confirmation purposes ... where, again, `ShowAddress` is identical to `GetAddress(show_display=true)`

in summary and after discussion, there doesn't currently seem to be a practical problem to be solved here, and sending `GetAddress` twice is the preferred solution, provided that processing the `GetAddress` message is fast.

closed 2020-02-21 by matejcik