trezor/trezor-firmware #868: Do not ask to format SD card when unlocking sd-protect

matejcik, 2020-02-21:
When the user tries to unlock Trezor with an unformatted SD card, and sd-protect is enabled, Trezor will first ask to format the card, and only then reject it as the wrong one.

Instead, it should reject the SD card straight away; an unformatted SD card is clearly not the correct one for sd-protect.

matejcik, 2020-02-21:
QA scenarios:

* enable SD protect and insert a correct SD card / an incorrect but formatted SD card / an incorrect and not-formatted SD card (make a NTFS filesystem on it or something)
* SD protect should fail the same way with a wrong card and with an unformatted card

closed 2020-02-24 by matejcik

bosomt, 2020-07-21:
**QA OK**

web dev suite master 8cc3650adbb123a224def6e0a024f25a8b7f5cc8
fw 2.3.2 e70811ce
latest nixos and chromium

PR #869 (2020-02-24): Do not ask for formatting in SD-protect checks
Fixes #868. Happens to fix a somewhat nasty problem in the previous `get_filesystem` code too.
commit d7b95823 (2020-02-24): core/sdcard: add ensure_filesystem option (fixes #868)

It is possible to call `ensure_sdcard` in a way that requires only SD
card be inserted, but not necessarily formatted.

This is useful for SD-protect and possibly other use-cases where the SD
card is read-only, and "not formatted" is identical to "not containing
the right files".