trezor/trezor-firmware #870: Error when cloning the repository on Windows 10

mark-stopka, 2020-02-22:
**Describe the bug**
When cloning trezor/trezor-firmware repository, following error occurs:
```
$ git clone https://github.com/trezor/trezor-firmware
Cloning into 'trezor-firmware'...
remote: Enumerating objects: 261, done.
remote: Counting objects: 100% (261/261), done.
remote: Compressing objects: 100% (141/141), done.
remote: Total 70062 (delta 154), reused 172 (delta 114), pack-reused 69801
Receiving objects: 100% (70062/70062), 32.08 MiB | 7.37 MiB/s, done.
Resolving deltas: 100% (51607/51607), done.
error: invalid path 'common/defs/ethereum/aux.png'
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

**Firmware version or revision**
`master`, revision 39ce100

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: N/A
 - OS:
```
PS C:\Users\mark_stopka> [System.Environment]::OSVersion.Version

Major  Minor  Build  Revision
-----  -----  -----  --------
10     0      18363  0
```
 - Browser: N/A
 - Version:
```
$ git --version
git version 2.25.1.windows.1
```
Nothing obvious would pop-up in git config:
```
[http]
        sslBackend = schannel
[diff "astextplain"]
        textconv = astextplain
[filter "lfs"]
        clean = git-lfs clean -- %f
        smudge = git-lfs smudge -- %f
        process = git-lfs filter-process
        required = true
[credential]
        helper = manager
[core]
        autocrlf = input
        fscache = true
        symlinks = true
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run `git clone https://github.com/trezor/trezor-firmware` on Windows 10
2. Observer error

**Expected behavior**
Repository would clone and check out to `master` branch on all Git supported platforms.

**Additional context**
Add any other context about the problem here.

mark-stopka, 2020-02-22:
When I downloaded the `master` zip file and browsed it, I've noticed file `aux.png` is present, however, when I've extracted the .zip file, the `aux.png` is not present, instead `_aux.png` exists. I removed the `aux.png` file in my [fork](https://github.com/2nd-Layer/trezor-firmware/tree/fix-upstream-issue-870/common/defs/ethereum) and I can now clone to this specific branch with `git clone --branch fix-upstream-issue-870 git@github.com:2nd-Layer/trezor-firmware.git`. However getting the file using `curl -L -O https://raw.githubusercontent.com/trezor/trezor-firmware/master/common/defs/ethereum/aux.png` and adding it back into the branch using `git add` breaks the branch on Windows again.

prusnak, 2020-02-22:
LOL, this is a relict from MS-DOS: https://en.wikipedia.org/wiki/DOS#Reserved_device_names

So it seems that even in Windows 10 you cannot have a filename containing one of the following: CON, AUX, PRN and LST.

mark-stopka, 2020-02-22:
LOL, that's funny, however `LST` works, `NUL` does not...

```powershell
PS C:\Users\mark_stopka> New-Item -Path .\AUX -ItemType File
New-Item : Could not find file '\\.\AUX'.
At line:1 char:1
+ New-Item -Path .\AUX -ItemType File
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : WriteError: (C:\Users\mark_stopka\AUX:String) [New-Item], FileNotFoundException
    + FullyQualifiedErrorId : NewItemIOError,Microsoft.PowerShell.Commands.NewItemCommand

PS C:\Users\mark_stopka> New-Item -Path .\CON -ItemType File
New-Item : FileStream was asked to open a device that was not a file. For support for devices like 'com1:' or 'lpt1:', call CreateFile, then use the FileStream constructors that take an
OS handle as an IntPtr.
At line:1 char:1
+ New-Item -Path .\CON -ItemType File
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [New-Item], NotSupportedException
    + FullyQualifiedErrorId : System.NotSupportedException,Microsoft.PowerShell.Commands.NewItemCommand

PS C:\Users\mark_stopka> New-Item -Path .\PRN -ItemType File
New-Item : Could not find file '\\.\PRN'.
At line:1 char:1
+ New-Item -Path .\PRN -ItemType File
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : WriteError: (C:\Users\mark_stopka\PRN:String) [New-Item], FileNotFoundException
    + FullyQualifiedErrorId : NewItemIOError,Microsoft.PowerShell.Commands.NewItemCommand

PS C:\Users\mark_stopka> New-Item -Path .\LST -ItemType File

    Directory: C:\Users\mark_stopka

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
-a----        22-Feb-20     18:21              0 LST

PS C:\Users\mark_stopka>
```

commit ae752216 (2020-04-09): common: rename auxilium chain to 'auxi' to avoid bad Windows name (fixes #870)
closed 2020-04-09 by matejcik