trezor/trezor-firmware #877: FIDO2: credential management extension

prusnak, 2020-02-26:
In order to fully support resident keys with OpenSSH we need to implement credential management extension

Let's evaluate whether this is something we want to do.

andrewkozlik, 2020-03-02:
AFAIK, the credential management in CTAP is not documented anywhere. However, from looking at the libfido2 code, it should be pretty easy to figure out. I already have a general idea of how it works:

Looks like there are several subcommands to the GetInfo command:
CRED_METADATA = 0x01
RP_BEGIN = 0x02
RP_NEXT = 0x03
RK_BEGIN = 0x04
RK_NEXT = 0x05
DELETE_CRED = 0x06

The RK_BEGIN subcommand has one parameter:
1: RP ID hash

The DELETE_CRED subcommand has one parameter:
1: Credential ID

Other subcommands don't have a parameter.

CRED_METADATA response:
1: Total number of resident credentials currently present on the authenticator.
2: Number of remaining resident credentials which can be created.

RP_BEGIN / RP_NEXT response:
3: PublicKeyCredentialRpEntity (map with keys "id" and "name") 
4: rp_id_hash (bytes)
5: Number of relying parties (RP_BEGIN only)

RK_BEGIN / RK_NEXT response:
6: PublicKeyCredentialUserEntity (map with keys "name", "displayName", "id")
7: PublicKeyCredentialDescriptors (map with key "id")
8: COSE Public Key (map with "alg", "crv" etc.)
9: Number of resident keys for the requested RP (RK_BEGIN only)

andrewkozlik, 2020-04-21:
Specification: https://fidoalliance.org/specs/fido2/fido-client-to-authenticator-protocol-v2.1-rd-20191217.html#authenticatorCredentialManagement

zejdlikt, 2020-12-18:
Hello, is there any decision already made on this topic? If so, any eta? 

Thank you

andrewkozlik, 2020-12-21:
Another version of the CTAP proposal is out: https://fidoalliance.org/specs/fido-v2.1-rd-20201208/fido-client-to-authenticator-protocol-v2.1-rd-20201208.html#authenticatorCredentialManagement. It's still in draft form, but I think this particular part of the specification if fairly stable by now and we can think about implementing it.

@zejdlikt, you can still used OpenSSH without this feature, the only trouble is if you are using multiple machines, then you need each one to have a copy of the "secret key" file in ~/.ssh/ (when using FIDO2 I think it doesn't actually contain any secret if I recall correctly, just metadata about the credential). Personally I am under the impression that this is a shortcoming of the design decisions around the OpenSSH implementation, because the credential management extension shouldn't be required for logging in with resident credentials, but I don't see deep enough into their implementation and design rationale to give justified criticism.

Anyway, I will look into how difficult this would be to implement.

zejdlikt, 2020-12-22:
> @zejdlikt, you can still used OpenSSH without this feature, the only trouble is if you are using multiple machines, then you need each one to have a copy of the "secret key" file in ~/.ssh/ (when [...]

Yes i know about the "workaround" there considering copying metadata in `~/.ssh/` - and of course it works (tested with TrezorT/Yubikey). Anyway, idea of having just a usb key with me traveling around the world and being able to login to my servers from any other server w/o `cp` or `scp` anything additional is a nice idea. So i would say, it would be very "nice to have" (if possible).

cs224, 2024-03-24:
Is there any update on this issue? Thanks!

Hannsek, 2024-03-26:
No progress has been done on this.

kagehisa, 2024-12-22:
Hi, are there any news on this issue? I'm using a Trezor Model T and still get a 
> Key enrollment failed: requested feature not supported
after tying
`ssh-keygen -vvv -t ed25519-sk -O resident -O verify-required`

Using SSH resident keys with the ability to enter a pin on the device was the reason I bought a Trezor in the first place.

Hannsek, 2024-12-22:
Unfortunately no news here. No ETA either.

kagehisa, 2024-12-22:
Is there already a feature branch or something? I'm happy to help out where I can but doing it completely from scratch in an unfamiliar codebase would be a bit tough.

Hannsek, 2024-12-22:
@andrewkozlik or @matejcik do you know? ⬆️

matejcik, 2025-01-02:
i don't think we touched this yet, sorry

kagehisa, 2025-01-04:
Is this on the list? 
This would be a pretty useful feature since the Trezor T and above are the only FIDO capable devices with pin entry on device. A real authenticator to go.

If there is a way to help getting this feature on the road I'm more than wiilling to help.

andrewkozlik, 2025-01-06:
I have to say that there are several things that make me hesitant to delve into this feature:
1. It's one thing to implement CTAP credential management in Trezor. However, to be spec compliant we would probably need to support CTAP 2.1, which may mean implementing a whole set of other features and changes in behavior, thus opening a nasty can of worms for development, review and extensive testing. On the other hand we could just add support for credential management and still claim that Trezor only support CTAP 2.0, hoping that it won't break or backfire at some point, since CTAP 2.0 does not specify credential management commands.
2. It really makes no sense to me why credential management is needed by OpenSSH in the first place. In fact it doesn't even make sense to me why resident credentials are needed by OpenSSH. Firstly, the private key file `id_ed25519_sk` contains primarily the FIDO credential, which is nothing secret and should logically be placed in `id_ed25519_sk.pub` which resides on the remote server. This way FIDO authenticators wouldn't even need to support resident credentials to function with OpenSSH. Secondly, assuming that there is some kind of rational reason not to place the FIDO credential into `id_ed25519_sk.pub`, credential management commands should still not be needed. I see no justifiable reason why to load resident keys from a FIDO authenticator to the SSH client computer, which is what `ssh-add -K` does. The normal way to work with resident credentials is to specify the RP ID `ssh:` in the authentication request to the authenticator. This all works well with CTAP 2.0. Frankly, I don't see why Trezor should be making up for apparently irrational design choices made in other software.
3. Credential management is a very niche feature that is not needed anywhere else other than this one very odd instance. Similarly we lack any other motivation to implement CTAP 2.1, since our CTAP 2.0 implementation currently provides all desired features outside of this issue.

TL;DR IMHO OpenSSH should store the FIDO credential in the public key file `id_ed25519_sk.pub`.

andrewkozlik, 2025-01-06:
An idea for a workaround: add a command to `trezorctl` that generates the private key file just like `ssh-add -K`.

prusnak, 2025-01-06:
> 2\. It really makes no sense to me why credential management is needed by OpenSSH in the first place.

Good observation ! I raised your comment on the [openssh-unix-dev](https://lists.mindrot.org/pipermail/openssh-unix-dev/2025-January/041739.html) mailing list. I will copy the relevant answers here once there are some.

kagehisa, 2025-01-06:
> It really makes no sense to me why credential management is needed by OpenSSH in the first place. In fact it doesn't even make sense to me why resident credentials are needed by OpenSSH. Firstly, [...]

I agree, the id file basically brakes every thirdparty ssh using automatism that is used to handle a set of private and public keys. 

> I don't see why Trezor should be making up for apparently irrational design choices made in other software.

Ok, I see your point but we are not talking about some random niche software. And Trezor uses the SSH key feature in its advertising. So some potential is seen here.
And most if not all other Wallets/Authenticators that have the "badge" FIDO2 are supporting this feature.

> An idea for a workaround: add a command to trezorctl that generates the private key file just like ssh-add -K.

This would require `trezorctl` on every host that the authenticator should be used. I see no benefit to manually carry over the id file and use ssh out of the box (if trezor is only used as authenticator). 

But I take it that this is no easy change and would require major effort to implement in the Trezor firmware, which will most likely not happen based on your comment.

andrewkozlik, 2025-01-07:
I found that when generating a FIDO resident key with OpenSSH 8.4 or newer, OpenSSH requires the authenticator to support the FIDO 2.1 credProtect extension. Meaning that resident credentials can be created on Trezor only using OpenSSH 8.2 and 8.3.

> An idea for a workaround: add a command to `trezorctl` that generates the private key file just like `ssh-add -K`.

To work with OpenSSH 8.4 and newer Trezor would also need to implement importing of credentials from the private key file into Trezor as a resident credential. Sounds like it's getting too complicated.

prusnak, 2025-01-10:
Got an answer from Damien Miller (OpenSSH maintainer).

Quoting from https://lists.mindrot.org/pipermail/openssh-unix-dev/2025-January/041744.html

Why FIDO key is stored on the client, not sever:

> It's done this way to allow FIDO keys to work along other key types in
the SSH protocol without requiring a whole new authentication subprotocol
just for FIDO keys, which would be needed to communicate the key handle
from the server back to the client - no such ability exists in the
existing SSH publickey authentication subprotocol.
>
> This was a deliberate tradeoff we made when integrating FIDO keys
into SSH: making the integration more SSH-like (dramatically lowering
the costs for other SSH server vendors to add authentication-only
FIDO support) vs making it more web-like (lowering the costs of of
implementing the client side of generating assertions).

Why RPID is needed:

> First, the client doesn't a priori know which RPID to use, it might be
something other than "ssh:". That's one of the reasons why we store it
in the pub/privkey file. How does a user of a RK-capable token get this
information from a token? For OpenSSH, we implemented it as `ssh-add -K`
>
> I have no idea whether invoking keys by RPID alone works with other
RK-supporting FIDO 2 tokens, we've always provided the key handle + RPID
and no other FIDO 2 token vendor that I'm aware of has had trouble with
this.

kagehisa, 2025-01-15:
So in the end it comes back to

> In order to fully support resident keys with OpenSSH we need to implement credential management extension
> 
> Let's evaluate whether this is something we want to do.

Since I would be no help (except for testing) in implementing I guess this is a decision for SatoshiLabs .

matejcik, 2025-01-16:
general note: we may want the official Credential Management extension to replace our custom wire-protocol credential management. that way any FIDO management app on the host PC can work with Trezor out of the box.

unfortunately the official credential management extension does not support backing up the currently stored credentials :( so we'd still want our own commands for that. only perhaps if we could shove the encrypted representation into the metadata under some vendor field