trezor/trezor-firmware #882: When checking address on device it says enter passphrase and dialogue disappears

bosomt, 2020-02-28:
**Describe the bug**
When checking address on device in receive dialogue it says enter passphrase on device dialogue that will immediately disappear and then it will show you address.

**Firmware version or revision**

2.3.0 / RC1 | 86b010a1

**Desktop/smartphone setup (please complete the following information):**
 - Wstage wallet
 - nixos
 - chromium
- Version 80

**To Reproduce**
Steps to reproduce the behavior:
1. Go to receive tab
2. Click on Show full address
3. look at device display
4. See error

**Expected behavior**
It must show address and not dialogue and then address

bosomt, 2020-02-28:
update: same behavior with T1 1.9.0
looks more like stage wallet bug

tsusanka, 2020-02-28:
Nice issue!

Indeed this looks like a wallet bug. @mroz22 I am receiving Initialize with `session id = NULL` for every new "Click on Show full address". The correct passphrase is then automatically send from Wallet, but it's causing this ugly behaviour. Can you look into it? The correct solution is to send the session id received in Features in every subsequent Initialize.

tsusanka, 2020-02-28:
For some reason I am not able to trasnfer this to mytrezor (I guess different orgs) so I am leaving here.

bosomt, 2020-02-28:
same for exchange tab
electrum works fine

tsusanka, 2020-03-20:
@bosomt could you retest please? It seems fixed to me on current stage wallet.

bosomt, 2020-03-20:
fixed
works fine with 1.9.0 and 2.3.0 in current stage wallet

closed 2020-03-20 by bosomt

bosomt, 2020-07-20:
**QA OK**

FW 2.3.2 revision  e70811c
wallet b3f18eb73765f92dd35eff2c562ef1edb5dd9fa3
suite https://github.com/trezor/trezor-suite/commit/582ebd2f9c1b733058df3e6adf70f842c7ef394d

latest chrome and nixos