trezor/trezor-firmware #883: Passphrase entry dialog stays forever

tsusanka, 2020-02-28:
While hunting for #882 I have noticed another bug. The "Passphrase entry" dialog stays until another screen is shown, which is very confusing to the user.

matejcik, 2020-02-28:
> The dialog should disappear on PassphraseAck.

the dialog should disappear at the end of the flow, actually, when the homescreen reappears. How did you trigger this?

tsusanka, 2020-02-28:
Just go to https://suite.corp.sldev.cz/suite-web/develop/, enable passphrase and enter some passphrase.

I am just investigating.

matejcik, 2020-02-28:
could not reproduce. The text goes away as soon as I enter the passphrase, replaced by a "please wait" empty screen -- as it should, because "please wait" is deriving the seed

tsusanka, 2020-02-28:
That is really weird. I am getting the same behaviour also using `trezorctl get-address`.

tsusanka, 2020-02-28:
This happens on emulator with disabled animations only (`-a`).

closed 2020-02-28 by matejcik

tsusanka, 2020-02-28:
@matejcik ACKed. Did you mark the cardano ui tests for some reason or they were just redundant?

matejcik, 2020-02-28:
There's nothing going on in the test, it's homescreen -> type password -> homescreen. They should have been excluded before.

commit 7f91bc57 (2020-02-28): core: ensure default task restarts after a passphrase workflow (fixes #883)