trezor/trezor-firmware #888: Multisig get address failure message

tsusanka, 2020-03-03:
Calling GetAddress with this data:

```
{
    "address_n": [
        2147483693, 
        0, 
        0
    ], 
    "multisig": {
        "pubkeys": [
            {
                "node": "xpub6AF46XPkbc77t8w74QNXg9SshYxSutwkpvEg3jGKfMEUT74MJ5C689n7xnF1XfrvsMof8rh7sZPxH68i9rPom17CRyqbu56Semn9vDS79Eu", 
                "address_n": [
                    2147483693, 
                    0, 
                    0, 
                    0
                ]
            }, 
            {
                "node": "xpub6F9ucJkut2FDrpvEGYBq59Bnfii1Jb52e9R6covwNvUkdCo18FGhRBvM8VaiCwBVzdKJGXwbY8HwJiw4uhnYi3oKnB3dxNZ3dELJASXrVLe", 
                "address_n": [
                    2147483693, 
                    0, 
                    1, 
                    0
                ]
            }, 
            {
                "node": "xpub6DymKmgGGubzsiFu7w6hrqC7YgfaCMQ6v4zhPUbgQTvUv82AUp69ZRPeUkXQZ4rgXnYJMYJTYRTqfMfyE2SmgvMHbaVeMt3jE8g2u5W2d6g", 
                "address_n": [
                    2147483693, 
                    0, 
                    2, 
                    0
                ]
            }
        ], 
        "m": 2
    }, 
    "script_type": "SPENDMULTISIG", 
    "show_display": true
}
```

returns a firmware error:

```
File "apps/wallet/sign_tx/multisig.py", line 89, in multisig_get_pubkey
ValueError: Failed to derive
```

Let's investigate if this is a bug and if it is not whether we can return a more meaningful message to the user (or we can't due to security concerns).

prusnak, 2020-03-03:
Is there at least of the XPUBs owned by the device?

matejcik, 2020-06-22:
the `address_n` field in `multisig.pubkeys` is supposed to have a partial path, e.g., `m/<xpub>/0/0`. This way Trezor can derive the appropriate public key from the "account xpub".

The "failed to derive" error is coming from an attempt to do a hardened derivation (2147483693 is `45'`) on a public key. So this is invalid data for the multisig struct.

closed 2020-06-22 by matejcik