trezor/trezor-firmware #890: SD-protect - Screen freeze after replacing with ill-formatted SD card

sorooris, 2020-03-04:
**Firmware version or revision**
2.3.0 / RC1 86b010a1
Preconditions: at least 2 SD cards are needed

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: not needed
 - OS: nixos
 - trezorctl 0.12 (sd-protect feature)

**To Reproduce**
1. Set up SD card protection on the device (via trezorctl) with SD card # 1,
2. Eject SD card # 1 and format it to NTFS, 
3. Replace with SD card #2 and unlock with the PIN (either SD card protection screen - Wrong SD card or SD card problem needs to be displayed),
4. Do not close the screen and replace with SD card # 1.

**Expected behavior**
User is warned that the sd-card used is in incorrect format and won't be able to unlock the device.

**Actual behaviour**
Device screen freezes when the first (NTFS-formatted) SD card is inserted.

**Additional context**
A massive red warning could be displayed when setting up the sd-protect function to emphasize the importance of not losing or formatting the SD card - up to discussion.

sorooris, 2020-03-04:
#807, #833

matejcik, 2020-03-05:
This is most likely a TT hardware issue - the device cannot safely hot-swap a card. Crash or freeze is a likely result.

The issue will be somewhat mitigated by #892, which shows a dialog telling you to unplug and then insert a different card.

closed 2020-03-05 by matejcik