trezor/trezor-firmware #892: SD-protect: reject unformatted card as "Wrong SD card"

matejcik, 2020-03-05:
follow-up to #868 , already discussed [here](https://github.com/trezor/trezor-firmware/pull/869#discussion_r383218077)

**steps:**
1. enable SD-protect
2. insert a different unformatted (or NTFS-formatted) card into Trezor
3. plug in Trezor and click Unlock

**expected:**
the relevant SD-protect file is clearly not on the SD card, so it should end with "Wrong SD card" dialog:
> **Please unplug the device** and insert the correct SD card. [Close]

**actual:**
Currently Trezor cannot differentiate between different errors coming out of the `mount()` attempt. The generic "SD card problem" dialog is shown:
> There was a problem accessing the SD card. **[Retry]**/[Abort]

There is no point in retrying if the problem was a missing filesystem.

Moreover, TT hardware does not handle hot-swapping of SD cards, so the user might be tempted to remove the card and insert another, which will crash or freeze the device.

**solution:**

`trezorio.fatfs` should expose error code constants from ff.c, and use them as `OSError` error codes. 

if that is done, it is possible to write code like:
```python
try:
   load_sd_salt()
except OSError as e:
   if e.args[0] == fatfs.FR_NO_FILSYSTEM:
       # show wrong SD card dialog
   else:
       # show generic dialog
```
or even better, have a specialized subclass of `OSError` for this condition

matejcik, 2020-03-20:
QA scenario:

Set up sd-protect with a SD card
Unplug Trezor, and insert a NTFS formatted card.
Plug in Trezor, enter PIN.
Verify that the error shown is "Wrong SD card", without option to retry - and _not_ "problem accessing SD card", with option to retry

closed 2020-03-20 by matejcik

bosomt, 2020-03-23:
tester, it says Wrong SD card after i changed it to NTFS
after new FW, new seed and protection enable card was formatted and it works fine

sorooris, 2020-03-23:
Tested with trezorctl 0.12
Confirming, QA OK.

PR #893 (2020-03-20): SD-protect - properly show "Wrong card"
fixes #892