trezor/trezor-firmware #894: test suite timeouts fail with on-device tests

matejcik, 2020-03-06:
The test suite is currently set up to have a time limit of 60 seconds per test case. This is to prevent the CI being stuck when for some reason a testcase freezes the emulator.

The timeout is unsuitable for running on real hardware, because the device is much slower.

It is possible to override the timeout from the command line or via an environment variable, but this will not work for test cases that are explicitly marked with a bigger timeout.

---

It seems that using the timeouts this way isn't appropriate.

A potential solution is to not configure any timeouts in the test suite itself, and set up a timeout via an argument/envvar just in the CI. Local runs would not have this timeout, but that should not matter, as in a local run the developer can notice that the suite is stuck and shut it down by hand. Alternately anyone can set up an envvar for their liking.

A drawback of this solution is that we have tests that take longer than others. While 60 seconds is a reasonable timeout for "normal" tests, certain others (signing 100-input tx; testing 16-of-16-of-16 Shamir recovery) might need a longer time limit even in the emulator.

One option is to figure out a time limit appropriate for all tests. At a preliminary guess, 5 minutes per test case should be enough for the long test to finish, and not blocking the CI for too long in the normal cases.

Another option is to extend use of `pytest.mark.slow` to all relevant test, and have one test run with `pytest -m "not slow" --timeout=60` and another with `pytest -m slow --timeout=600`.

For now I'd go with the first option, setting 5 minutes in CI and seeing if that works OK.

cc @onvej-sl

onvej-sl, 2020-03-06:
Another option is to define a decorator that sets the timeout for a particular test relatively to the global timeout.

matejcik, 2020-03-06:
that is unfortunately not a feature of [pytest-timeout](https://pypi.org/project/pytest-timeout/) which we use for this, and I don't think it's worth developing for this usecase

tsusanka, 2020-03-06:
I am okay with all the options, just want to throw another one in: how about we apply the timeouts only if `--timeout` is provided? That way on default we have no limits and on CI we trigger it with this option.

matejcik, 2020-03-06:
again, that is not a feature of pytest-timeout :)

matejcik, 2020-03-09:
timeouts outside CI disabled in 5b7d11e4ee5d1f6559b4b9991b47a6dd006b663f, which should fix the issue

CI-wide timeout of 5 minutes configured in 3869a05677968e532272b68558f87dd210df04fa, we will see if this solution is sufficient

closed 2020-03-09 by matejcik