trezor/trezor-firmware #903: DataError: string overflow with T1 when PIN too long

bosomt, 2020-03-13:
**Describe the bug**
When you set wipe code or PIN code via trezorctl, you can enter long code and it will result with error **trezorlib.exceptions.TrezorFailure: DataError: string overflow**

**Firmware version or revision**
Firmware version 1.9.0 

**Desktop/smartphone setup (please complete the following information):**
 trezorctl 12 and 11.6

**To Reproduce**
trezorctl set pin
trezorctl set wipe-code
and enter longer number than max PIN lenght

**Expected behavior**
You will get proper error

bosomt, 2020-03-13:
Device gets wiped during this operation:

trezorctl set wipe-code
WebUSB transport is disabled: libusb-1.0.so: cannot open shared object file: No such file or directory
Please confirm action on your Trezor device
Use the numeric keypad to describe number positions. The layout is:
    7 8 9
    4 5 6
    1 2 3
Please enter current PIN: 

when I enter 90 device gets wiped,
can this be related or is it different issue ?
@matejcik @tsusanka

matejcik, 2020-03-13:
different issue, please create a separate one

bosomt, 2020-03-13:
sent https://github.com/trezor/trezor-firmware/issues/904

matejcik, 2020-03-20:
fixed in #906

closed 2020-03-20 by matejcik

tsusanka, 2020-03-20:
@matejcik does this need QA?

matejcik, 2020-03-20:
i don't think so, if #904 is tested