trezor/trezor-firmware #904: Device with 1.9.0 FW gets wiped when you enter PIN code with 0

bosomt, 2020-03-13:
**Describe the bug**
Device with 1.9.0 FW gets wiped when you enter PIN code with 0

**Firmware version or revision**
1.9.0

**Desktop/smartphone setup (please complete the following information):**
trezorctl 12.0

**To Reproduce**
Steps to reproduce the behavior:
1. create new wallet
2. set PIN 
3. trezorctl set wipe-code
4. enter 90 when asked for PIN code
5. device gets wiped

**Expected behavior**
device don't get wiped

closed 2020-03-17 by andrewkozlik

tsusanka, 2020-03-20:
@andrewkozlik this should be retested shouldn't it? Please don't forget the `needs QA` label next time.

andrewkozlik, 2020-03-20:
> @andrewkozlik this should be retested shouldn't it?

Definitely should.

bosomt, 2020-03-23:
tested with latest trezorctl and version:	2.3.0 RC1	revision:	6478b270

works FINE

@matejcik don't forget to release new trezorctl before 1.9.0 gets shipped

sorooris, 2020-03-27:
1.9.0 6478b270
2.3.0 8ae0535e
QA OK

PR #906 (2020-03-17): Check for invalid PIN
Fixes #904.