trezor/trezor-firmware #905: Cannot use the Trezor Wallet

ladislavmacoun, 2020-03-13:
**Describe the bug**

I haven't use my trezor for a while, now after wanting use wallet.trezor.io again I can't get my trezor working. I have installed the newest drivers which did not helped. The wallet.trezor.io keeps asking to install the bridge driver over again. I've disabled adblock, and tried to use the chrome app. Neither helped.

The deamon is probably not leaded correctly
```
sudo launchctl list | grep -i trezor
echo $?
1
```

On Mac, I can see the Trezor at the "Check for devices", on windows it is not even listed.

**Firmware version or revision**
Bridge Driver 2.0.27

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Wallet
 - OS: Windows 10, macOS  10.14.6 and 10.15.3
 - Browser Chrome
 - Version 80.0.3987.132 (Official Build) (64-bit)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  https://wallet.trezor.io https://old-wallet.trezor.io
2. Click on https://wallet.trezor.io/#/bridge
3. Install the bridge software
4. Reopen 1.

**Expected behavior**
Install the bridge driver and get to wallet.

**Screenshots**
[image]
*Additional context**
Add any other context about the problem here.

ladislavmacoun, 2020-03-13:
I forgot to mention it's Trezor One

ladislavmacoun, 2020-03-13:
I tried to use Firefox instead of Chrome and it works, however, it's weird since the 2fa is working on chrome. 🤷‍♂️

prusnak, 2020-03-13:
Since it's working for you in Firefox this is hardly a firmware issue. If you need further help contact our support please.

closed 2020-03-13 by prusnak