trezor/trezor-firmware #910: Drop decred_script_version

matejcik, 2020-03-17:
`TxInputType`, `TxOutputType` and `TxOutputBinType` all have a `decred_script_version` field.

We need that field in `TxOutputBinType`, to correctly calculate prevhash.
However, we do not need it in `TxInputType` nor `TxOutputType`, because we accept only v0 anyway.
That suggests that the `decred_script_version` field could be removed from the latter messages, and 0 would be assumed by code.

Out of interest, do we know if other versions than 0 are used in the wild? If yes, might we (ever) want to add support for them?

tsusanka, 2020-03-18:
In another commit we have removed this check:

```
if (coin->decred && tx->bin_outputs[0].decred_script_version > 0) {
      fsm_sendFailure(
           FailureType_Failure_DataError,
              _("Decred script version does not match previous output"));
      signing_abort();
      return;
}
```
which somewhat implies we might have not supported bin_outputs with `decred_script_version != 0` anyway. We have removed similar check for inputs as well, but since this issue removes that it is of no concern.

In this issue we have to decide whether we should remove it from the OutputBin as well or we should restore the check.

tsusanka, 2020-04-23:
Waits for #617

PR #1002 (2020-05-21): Drop Decred script_version
Closes #910.
closed 2020-05-21 by tsusanka