trezor/trezor-firmware #912: explicit closing of sessions

matejcik, 2020-03-23:
With the newly introduced multi-session support, there are some usecases where we want to use a session for a short time and then free up the cache slot for further uses.

* when we want to reprompt for a passphrase (e.g. in Trezor Suite empty wallet confirmation), we need to allocate a new session. Every reprompt potentially evicts one of the valid passphrased sessions.
* when enumerating connected devices, we should (although don't _need_ to, technically speaking) initialize a session to get Features. repeated enumerations would also needlessly evict sessions

We propose a new message, `EndSession`, that cancels the current session, erases its ID and removes it from the cache.
Subsequent calls without `Initialize` would return a new failure code, `InvalidSession`. For ease of implementation, we could allow calls that do not touch session data (e.g., device settings), and return the failure only for calls that use the seed.

closed 2020-08-28 by matejcik

tsusanka, 2020-09-25:
This probably doesn't need QA since it is not yet in Suite right?

matejcik, 2020-09-25:
correct

PR #1227 (2020-08-28): EndSession support
this PR introduces an implementation of the `EndSession` message and fixes #912 

the new behavior is as follows:

- until first `Initialize` message is received, any attempt to use cached data is rejected with a `ProcessError("Invalid session")`
- after `Initialize`, a session is allocated
- calling `EndSession` destroys data of that session, and resets to the "invalid session" state. Subsequent uses of cache will again raise a `ProcessError`

This allows us to reuse session cache slots: a session ended with `EndSession` is cleared away, so we can have 9 "long-lived" sessio [... cut]