trezor/trezor-firmware #913: X button during onboarding skips confirmation

bosomt, 2020-03-24:
**Describe the bug**
X button during onboarding skips confirmation

**Firmware version or revision**
version:	2.3.0 RC1	revision:	6478b270

**Desktop/smartphone setup (please complete the following information):**
stage wallet in latest chrome

**To Reproduce**

1) start device backup
2) show 12 words
3) you have two options X or Hold to confirm
4) confirm button confirms, x button skips to word confirmation

**Expected behavior**
I don't know what X button should do, abort seed backup ? Is it even possible when you already seen it.

Maybe we should hide it. 

**Screenshots**

[image]

tsusanka, 2020-03-24:
Yes, the X button should cancel the progress, it was implemented in https://github.com/trezor/trezor-firmware/issues/76. Are you saying that in this case the backup is not aborted but the progress continues with the words checking?

bosomt, 2020-03-24:
@tsusanka yes exactly, when you hit X it will ask you to enter random 3 words, same behaviour as hold button

tsusanka, 2020-03-24:
Good catch. Will look into it.

bosomt, 2020-03-24:
same behavior with shamir backup

tsusanka, 2020-03-24:
Fixed in aa71c20f (to be released).

matejcik, 2020-03-24:
this is exactly the place where the "X" button makes no sense, no?

bosomt, 2020-03-30:
works fine now. but there is some inconsistency

but when I press X in Shamir new wallet it looks like this:

[image]

and with single seed it looks like this:

[image]

sorooris, 2020-03-30:
QA NOK, revision 8ae0535e
stage-wallet.trezor.io commit 9d9d8de  

Actual behavior: Selecting the X button while creating a single seed backup aborts the backup and fails it; wallet prompts the user to wipe the device and start again.

Expected behavior: Either selecting the X button soft-fails the backup process (process can be initialized from the Wallet again) or the X button is not present in the device at all.

tsusanka, 2020-03-30:
I'll look into it.

tsusanka, 2020-03-30:
I agree this is a bit confusing, but after some digging I do not think this is a bug. Let's describe the workflow with respect to unpluging the device:

If you unplug Trezor during:

- Single Backup, it marks the backup as invalid. The Single Backup is _delayed_ by default.
- Shamir DIRECT Backup, it **does not** mark the backup as invalid.
- Shamir DELAYED Backup, it marks the backup as invalid.

By direct/delayed backup I mean if you decide to do the backup right away during the initialization or later using the backup button in Wallet. Remember, Shamir has a screen for this, which Single Backup does not (though it should?):

[image]

I believe this behaviour is correct, because in DIRECT Shamir backup the seed is written to storage _after_ the backup is successful. So you can freely do another initialization.

cc @matejcik 

-----

It is true though that the screen above lies a little - the wallet is not actually created, it gets created after the backup "Skip" is chosen or after the backup. Maybe the correct solution is indeed to store the seed right away and then mark the backup as invalid in case of Trezor unplug? But it sounds less user-friendly. One way or another it sounds as another issue (which I open).

-----

The X button only mimics this behaviour. So I believe this is correct. We might consider @matejcik's thought above though - whether it makes sense to have the X button here. Maybe we should indeed get rid of it here so it does not confuse users, maybe some might think that it is a way to go "back"?

matejcik, 2020-03-30:
quoting from @bosomt's original report:

> **Expected behavior**
> I don't know what X button should do, abort seed backup ?
> Is it even possible when you already seen it.
> There is no button X in 2.1.8 FW
> 
> Maybe we should hide it.

I think this is actually a strong argument for removing the X button: it's unclear what should be expected to happen when you click it.

---

@tsusanka how sure are you about this?

> Remember, Shamir has a screen for this, which Single Backup does not (though it should?)

reading [the code](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/management/reset_device/__init__.py#L66-L68) ISTM the same screen exists with Single backup -- and, same as with Shamir direct, Single direct would also not write the seed to storage if you unplug

tsusanka, 2020-03-30:
> I think this is actually a strong argument for removing the X button: it's unclear what should be expected to happen when you click it.

I agree. I was preparing it in the meantime, I will push soon.

> reading the code ISTM the same screen exists with Single backup -- and, same as with Shamir direct, Single direct would also not write the seed to storage if you unplug

Ahaa! It did seem weird to me! It is a Wallet inconsistency not firmware's! Wallet sends `msg.skip_backup = True` for Single Backup and `msg.skip_backup = False` for Shamir Backup. I will check if this is true also for Suite and will open an issue appropriately.

tsusanka, 2020-03-30:
@bosomt @sorooris Firmware in email. Please test the X button is no longer present during the reset.

bosomt, 2020-03-31:
@tsusanka works fine for me, i will send inconsistency when creating wallet to wallet repo

bosomt, 2020-03-31:
@tsusanka ok i see that you already did it ;)

closed 2020-03-31 by tsusanka

sorooris, 2020-04-01:
QA OK (2.3.0, revision: 0b7a8449)