trezor/trezor-firmware #920: Unable to verify firmware

yura-pakhuchiy, 2020-03-26:
I've tried building firmware v2.1.8 using docker and fingerprint do not match with the official release. I've used the following commands:
```
$ git checkout core/v2.1.8
$ git submodule update --init --recursive
$ PRODUCTION=1 ./build-docker.sh core/v2.1.8
...
tools/binctl build/firmware/firmware.bin -h
Trezor Vendor Header
  * magic   : TRZV
  * hdrlen  : 4608
  * expiry  : 0
  * version : 0.0
  * scheme  : 2 out of 3
  * trust   : 65535 = [. . . . . . . . . . . . . . . .] = []
  * vpub #1 : 47fbdc84d8abef44fe6abde8f87b6ead821b7082ec63b9f7cc33dc53bf6c708d
  * vpub #2 : 9af22a52ab47a93091403612b3d6731a2dfef8a33383048ed7556a20e8b03c81
  * vpub #3 : 2218c25f8ba70c82eba8ed6a321df209c0a7643d014f33bf9317846f62923830
  * vstr    : SatoshiLabs
  * vhash   : 0b47bb9e55673134f3d8e96bb379ec0411a4d8548feb8e52e6ffb45fdb749169
  * vimg    : (4403 bytes)
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : 78dd3cce48d58e880da32b279b331dbec78459520a711d5f6aef75e39626485d89480f77de74ddbc2fc2dd24e70c4d4cb8bd6d73c16d512a49f1a8d03ca5a10a
  * fngprnt : abb773af7aa4c475e505d6e1b462898bc7ebc7bc493e34d21fb2b47912d8f95b

No firmware signature
Trezor Firmware Image
  * magic   : TRZF
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 1548288
  * version : 2.1.8.0
  * fixver  : 2.1.0.0
  * hashes: OK
    - 00 : 22f690ae432a5cbbbc263dc1ed8db36f6faa994b91df850eaa43c0ef230063ed
    - 01 : 752cda20d50b0b2cd3a1867a8a29f590ccb1aa3dfc284e9e457dc6182f6de463
    - 02 : bf4aa5bb6a1bc52a1737870983071d8d633cd39a5d91caa704cce4270256411a
    - 03 : 1600122d8300ec4ed6e48f0e2fcfd98d0e7b7fc1e8dc628289908b1cc45a07aa
    - 04 : db45360150787f4f96ed802b3999b93b27b0053cfcff232bee6355bbb135e3d6
    - 05 : e4dc668bc62e8c219aa8efc7fa3642022287865f96576d45b93967d0da24e08c
    - 06 : 83157df5ddf6cc9a7a9438b678f8d3bd066556cee96de28b27e2bdac1214cac6
    - 07 : 8b4c13f71f898aab3c6f462b0984bb04014fd7bf06283e15d96b115f99b7cade
    - 08 : e2f313864afe822aa5d5ba566857fea54256fe46fb672dc00a4448de114097d0
    - 09 : 6b29f4c721630d72a83e6fdd933e29bd5300a6caa71c7308936309d70f9c926a
    - 10 : b72e6bb7572661aa21154d755d8cdc200c6e2142b18c115c7b7ff57c20bdb03e
    - 11 : 8f3845482f17d57d40343a92a627c116ee52fd3c83dd8145069c3c8250fb715a
    - 12 : 0000000000000000000000000000000000000000000000000000000000000000
    - 13 : 0000000000000000000000000000000000000000000000000000000000000000
    - 14 : 0000000000000000000000000000000000000000000000000000000000000000
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x00 = [. . . . . . . .]
  * sig     : 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  * total   : 1553920 bytes
  * fngprnt : d966f1a29c8cce85d598aca7eab2f47c34dc9cd9313e269e3a0194f04d804ce1
```
Fingerprint does not match with officially released file:
```
$ ./binctl ~/trezor-2.1.8.bin 
Trezor Vendor Header
  * magic   : TRZV
  * hdrlen  : 4608
  * expiry  : 0
  * version : 0.0
  * scheme  : 2 out of 3
  * trust   : 65535 = [. . . . . . . . . . . . . . . .] = []
  * vpub #1 : 47fbdc84d8abef44fe6abde8f87b6ead821b7082ec63b9f7cc33dc53bf6c708d
  * vpub #2 : 9af22a52ab47a93091403612b3d6731a2dfef8a33383048ed7556a20e8b03c81
  * vpub #3 : 2218c25f8ba70c82eba8ed6a321df209c0a7643d014f33bf9317846f62923830
  * vstr    : SatoshiLabs
  * vhash   : 0b47bb9e55673134f3d8e96bb379ec0411a4d8548feb8e52e6ffb45fdb749169
  * vimg    : (4403 bytes)
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : 78dd3cce48d58e880da32b279b331dbec78459520a711d5f6aef75e39626485d89480f77de74ddbc2fc2dd24e70c4d4cb8bd6d73c16d512a49f1a8d03ca5a10a
  * fngprnt : abb773af7aa4c475e505d6e1b462898bc7ebc7bc493e34d21fb2b47912d8f95b

Firmware signature OK
Trezor Firmware Image
  * magic   : TRZF
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 1548288
  * version : 2.1.8.0
  * fixver  : 2.1.0.0
  * hashes: OK
    - 00 : 22f690ae432a5cbbbc263dc1ed8db36f6faa994b91df850eaa43c0ef230063ed
    - 01 : 5fe4c20c83532a885377928dbfd6f8b15238ce630dc349133fc00db902d923f8
    - 02 : 42d03b274c3db75ceb7acbe11b9ed356cc1bcfd682352374316aee5a69ccc647
    - 03 : 42b3b0c718e0bb03c805340f9fbf812f8f2d97fcf1823a576c4bc20482325837
    - 04 : 502c6fcb036698c14ecadd4ab5cda1431b5ba07258458837b06d9803aae31310
    - 05 : e4dc668bc62e8c219aa8efc7fa3642022287865f96576d45b93967d0da24e08c
    - 06 : 83157df5ddf6cc9a7a9438b678f8d3bd066556cee96de28b27e2bdac1214cac6
    - 07 : 8b4c13f71f898aab3c6f462b0984bb04014fd7bf06283e15d96b115f99b7cade
    - 08 : e2f313864afe822aa5d5ba566857fea54256fe46fb672dc00a4448de114097d0
    - 09 : 6b29f4c721630d72a83e6fdd933e29bd5300a6caa71c7308936309d70f9c926a
    - 10 : b72e6bb7572661aa21154d755d8cdc200c6e2142b18c115c7b7ff57c20bdb03e
    - 11 : 8f3845482f17d57d40343a92a627c116ee52fd3c83dd8145069c3c8250fb715a
    - 12 : 0000000000000000000000000000000000000000000000000000000000000000
    - 13 : 0000000000000000000000000000000000000000000000000000000000000000
    - 14 : 0000000000000000000000000000000000000000000000000000000000000000
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : c39f55429afbe23c806eea71bb703a9ba87c22a201793c492aae23a1c3fe0a32232465e7955c2ac4c0e328a15b6cada085e665ba58c1b55ab7be19f6043ac20e
  * total   : 1553920 bytes
  * fngprnt : 8a5fa12132651b6e33344fd025d0d90885f5cc1c342427ebcea4f0ae98b50d8c
```

I've also tried building 2.1.6 and 2.2.0, fingerprints did not match as well.

matejcik, 2020-03-26:
this is interesting, because only code hashes 1-4 are mismatched, so most of the firmware is identical

prusnak, 2020-03-26:
Something might have changed in the docker container, for example.

However, we don't guarantee deterministic builds being deterministic for eternity. Currently, we can only guarantee this is correct only at the time of the release.

This might be switched if we change the underlying distro to something less unpredictable than Debian - NixOS or Guix comes to mind.

yura-pakhuchiy, 2020-03-26:
Maybe just disabling repositories with packages updates and installing versions from initial distro release will do the trick?

yura-pakhuchiy, 2020-03-26:
> Currently, we can only guarantee this is correct only at the time of the release.

I do understand that deterministic build for eternity is not something easy to achieve. Though ability to verify the latest stable release before flashing it would be nice.

prusnak, 2020-04-01:
At this point I am afraid there is not much we can do. I opened another, more generic, issue #926 

Thanks for the report

closed 2020-04-01 by prusnak