trezor/trezor-firmware #921: Inconsitency in amount formating and buggy `bn_format`

onvej-sl, 2020-03-30:
The functions `format_amount(amount, decimals)` (in `core/src/trezor/utils.py`) and `bn_format(amount, NULL, NULL, decimals, False)` (in `crypto/bignum256.c`)  in my opinion should return the same string but they don't.
When `amount = 10` and `decimals = 1`, `format_amount` returns `"1"` whereas `bn_format` returns `"1.0"`.

The functions `stellar_format_price(numerator, denominator, out, outlen)` (in `legacy/firmware/stellar.c`) and `"%f" % numerator / denominator` (in `_confirm_offer` in `/core/src/stellar/operations/layout.py`) should return the same string but they don't.
When `numerator = 1` and `denominator = 1`, the former returns `1.0`, the latter returns `1.000000`

Moreover `bn_format(1, "", "s", 0, -1, out, outlen)` returns `"s"` instead of `"0s"`.

tsusanka, 2020-05-21:
@onvej-sl do I understand correctly, this is done now?

prusnak, 2020-05-22:
There are 3 issues in the OP. I think only the second one is fixed, not sure about the others.

onvej-sl, 2020-05-22:
The first and the third are fixed in https://github.com/satoshilabs/trezor-firmware/pull/78, so it's done.

closed 2020-05-22 by tsusanka

onvej-sl, 2020-05-22:
QA: Integer amounts on trezor one should be now formatted without the trailing `.0` (the same way they are on trezor T). That is `1` instead of `1.0`. Please, let me know if you notice an (integer or non-integer) amount on **trezor one** that is formatted wrong.

sorooris, 2020-06-02:
tested against https://trezor.io/stellar/
devices: T1 1.9.1 revision 81553df0

Compared to the latest public T1 release (fw 1.9.0) and the trailing `.0` is not displayed in Stellar transactions; tested several amounts.