trezor/trezor-firmware #922: buttons get stuck on passphrase keyboard

matejcik, 2020-03-30:
When typing passphrase on device, sometimes a button gets stuck in a highlighted state, and the corresponding letter is not typed.
Clicking again, or somewhere else, "unsticks" the button.

This happens both on real hardware and in the emulator.

It is as if the device registers a TOUCH_START but drops a TOUCH_END event. Sometimes the button does not highlight but the letter is typed, again, as if TOUCH_END was registered and the corresponding TOUCH_START dropped.

This does not seem to affect buttons in any other screen, so it is likely a race in the passphrase keyboard code.

matejcik, 2020-03-31:
:))

[image]

matejcik, 2020-03-31:
Hmm
So the fundamental problem is that two tasks are waiting for `io.TOUCH` interface: the side-swipe task for changing pages, and the task that clicks buttons.
The side-swipe task is faster to reschedule itself.
Sometimes the TOUCH_UP event comes in the short interval where side-swipe is already waiting for touch, but button-clicker isn't. Then the side-swipe receives and ignores the event, and button-clicker never sees it.

This doesn't seem fixable without (a) rethinking the event scheduler so that this is not a problem, or (b) merging side-swiper and button-clicker into the same task.

matejcik, 2020-04-01:
a draft solution to this problem makes `loop.race` "more transparent" in terms of scheduling; namely: any `Syscall` instances are executed within the same context as the `loop.race` syscall, and returns from these syscalls will clean up and resume directly to the originating context

that should ensure that `loop.race(loop.race(loop.race(...(io.wait())))` is identical, in terms of scheduler steps, to a plain `io.wait()`

This works fine for this and similar case. But it is unclear if the same problem can't occur again in a situation where the input handler is not fast enough for a reason other than `loop.race`

tsusanka, 2021-11-05:
Ping @matejcik. Will be solved by the new Rust layout UI.

matejcik, 2021-11-08:
Let's close this, because it's overwhelmingly likely that moving to Rust UI closes this issue -- and if a similar one exists, there is no reason to track it as the same thing.

Note that there is an XFAILed device test blocked on this, so when this is solved, we'll get a testsuite failure and we'll need to re-enable the test.

closed 2021-11-08 by matejcik

matejcik, 2023-01-31:
noting here that UI2 did indeed solve the problem. test was reenabled in #2778