trezor/trezor-firmware #923: Session handling on device for stateless software

achow101, 2020-03-31:
Currently, session handling is done by the software and requires the software to maintain state in order to have a good experience. My understanding is that sessions will be improving in the future. But it seems like the current direction is still towards having the software maintain some kind of session state. It would be preferable for such session handling to also be done on device so that the software does not need to maintain state. Otherwise stateless software such as HWI become increasingly frustrating to use.

matejcik, 2020-03-31:
The new firmwares allow you one of two options:
(a) Unlock the device via something like `trezorctl get-session` and store the session key as a token for further access. Because we can now handle multiple sessions simultaneously, the token has a very good chance of being valid for as long as the device is connected.
(b) Store passphrase on the host and send it to the device without user interaction, the way it used to work on Trezor One. Then you don't need to care about session lifetimes because you can create a short-lived session whenever you need it.

We are mostly moving in the direction of (a), with plans to have encrypted sessions at the USB level.

I'm not sure if "session handling on device" is a meaningful term, but my guess is that your usecase is something like:
1. user connects device
2. user unlocks device with a passphrase, which is transparent to the host
3. the device stays unlocked for all uses coming from the host

Am I correct?

We briefly considered allowing the host software to force a session identifier; i.e., `Initialize(session_id=0xfffffffff)` would create a valid session 0xffffffff. This was ultimately scrapped because "session id was not repeated back to me" is an indicator that the session was inactive.
We might revisit this, or maybe allocate a hard-coded session for "session-less" use. The drawback is that it would lead developers to use this instead of maintaining session state properly.

achow101, 2020-03-31:
> I'm not sure if "session handling on device" is a meaningful term, but my guess is that your usecase is something like:
>
>     1. user connects device
>
>     2. user unlocks device with a passphrase, [...]

Yes, that is what I'm looking for.

But I was also thinking of actual session management on the device itself. There could be an button that brings up a list of active sessions on that device and allows the user to change which session he wants to use. Or remove a session. Or add one. Basically the user can choose a session on the device instead of the software having to remember which session to use. Of course software could still remember which session to use and just do that too.

prusnak, 2020-03-31:
> But I was also thinking of actual session management on the device itself. There could be an button that brings up a list of active sessions on that device and allows the user to change which [...]

This sounds very complicated and user experience hell, honestly. It's much easier to add the management of one session to the client software than to add management of all active sessions to the hardware.

closed 2020-03-31 by prusnak

achow101, 2020-03-31:
> This sounds very complicated and user experience hell, honestly. It's much easier to add the management of one session to the client software (just remember a single integer) than to add management of all active sessions to the hardware.

With a stateless software, this is also user experience hell because the user has to keep re-entering his passphrase.

prusnak, 2020-04-01:
That's why I suggest you remember the session_id in your software, so the user does not have to keep re-entering it. Surely adding one variable to Python code is much less work than what you are suggesting.

matejcik, 2020-04-01:
hwi does not (seem to) store state between CLI invocations, so each operation = one invocation = passphrase entry

That is why `trezorctl get-session` exists. hwi would need to implement something similar, but it _is_ a piece of state to be managed.