trezor/trezor-firmware #926: Make deterministic build better

prusnak, 2020-04-01:
We are using Debian in Docker to have deterministic builds, but this is far from optimal (see #920 for example). There are other options to do this:

* gitian (old bitcoin method)
* guix (new bitcoin method)
* maybe nixos? (similar to the one above)

prusnak, 2020-04-01:
Related to https://github.com/trezor/trezor-firmware/issues/692

prusnak, 2020-06-07:
Switched to Nix in Docker via https://github.com/trezor/trezor-firmware/pull/1036

closed 2020-06-07 by prusnak

tsusanka, 2020-06-07:
Btw [this](https://github.com/trezor/trezor-firmware/commit/22533a3f31ba1e8d896a6b57e8002606d6eceefa) modification is needed if we want to build and test using Nix as well, which I guess we do.

The CI is [failing](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/153758897) with that commit. I can investigate tomorrow.

prusnak, 2020-06-07:
> The CI is failing with that commit. I can investigate tomorrow.

Argh, mea culpa. Pushed fixed in your branch - hopefully this fixes it.

tsusanka, 2020-06-08:
Almost there! Added `binutils` to fix `strip not found` error.

brianddk, 2022-02-10:
> Switched to Nix in Docker via [#1036](https://github.com/trezor/trezor-firmware/pull/1036)

@prusnak I'd still be interested in doing a PR for Guix if there is any buy-in from the maintainers?  It can be [run in docker if needed](https://github.com/pjotrp/guix-notes/blob/master/CONTAINERS.org).  Would there be any support for this, or is it too soon to look at redesigning the build environment?

prusnak, 2022-02-10:
> I'd still be interested in doing a PR for Guix if there is any buy-in from the maintainers?

We prefer to use Nix, not Guix, for our build system. One of the reasons is that some of Trezor people are active maintainers in Nixpkgs repository, while we don't have reach to Guix ecosystem. That means there will be no buy-in from our side at this moment.