trezor/trezor-firmware #927: Create new attestation certificate for WebAuthn

andrewkozlik, 2020-04-01:
The WebAuthn spec imposes some requirements on the attestation certificate (https://www.w3.org/TR/webauthn/#packed-attestation-cert-requirements), which Trezor is currently not complying with:
- Version: 3
- Subject-C: "CZ" (PrintableString)
- Subject-O: "SatoshiLabs, s.r.o." (UTF8String)
- Subject-OU: "Authenticator Attestation" (UTF8String)
- Subject-CN: UTF8String of our choice.
- BasicConstraints.CA = false
- The AAGUID in an extension.

I encountered this when I wanted to add a basic attestation test to fido2-tests, but found that it's not passing due to the wrong certificate version. So we should generate a new certificate in accordance with the above requirements.

Questions @stick @onvej-sl:
- What should the Subject-CN be? Currently we have "Trezor U2F". The options that come to mind are "Trezor 2" (the AAGUID is the hash of this string), "Trezor T" or "Trezor FIDO".
- Should we use this new certificate for U2F?
- Do we want to reuse the old private key?

prusnak, 2020-04-01:
my 2 cents:

1) Trezor FIDO
2) yes
3) yes

andrewkozlik, 2020-04-02:
I think (2) and (3) shouldn't break anything.

YubiKey uses "Yubico U2F EE Serial 61730834" as the common name. Not sure what the EE stands for. The number is the SN of the certificate. So we could use "Trezor FIDO Serial <SN>", where SN is a random 8-digit number.

BTW, I think BasicConstraints.CA=false means the certificate shouldn't be self-signed, so there should also be a root, e.g. "Trezor FIDO Root Serial <SN2>".

andrewkozlik, 2020-04-02:
Perhaps this:
-----BEGIN CERTIFICATE-----
MIIByjCCAXCgAwIBAgIEA0VgxDAKBggqhkjOPQQDAjAuMSwwKgYDVQQDDCNUcmV6
b3IgRklETyBSb290IENBIFNlcmlhbCA4NDE1MTM1NjAgFw0yMDA0MDIxMzU5MDNa
GA8yMDUwMDQwMjEzNTkwM1owdTELMAkGA1UEBhMCQ1oxHDAaBgNVBAoME1NhdG9z
aGlMYWJzLCBzLnIuby4xIjAgBgNVBAsMGUF1dGhlbnRpY2F0b3IgQXR0ZXN0YXRp
b24xJDAiBgNVBAMMG1RyZXpvciBGSURPIFNlcmlhbCA1NDg3ODQwNDBZMBMGByqG
SM49AgEGCCqGSM49AwEHA0IABNkYvfqKVKyS6Q2pH8p6omRUwNFzNjFN3oOlS4a1
307wUmWaHW/8t0Z/Gs3bijMIC17tkYkT9EOlJhvHe2hgb8GjMzAxMCEGCysGAQQB
guUcAQEEBBIEENbQvcNi7sTb3o16ZW5KRIcwDAYDVR0TAQH/BAIwADAKBggqhkjO
PQQDAgNIADBFAiAsBNDeBlNtW6boe+vgay6iHY7rKxMOG/237QQ2gCbVUwIhAMw5
PbCvOtaFybw1nTRVFbANY2igpvATVWNiUYVBoB32
-----END CERTIFICATE-----
```
SEQUENCE
    SEQUENCE
        cont 0
            INTEGER	2
        INTEGER	54878404
        SEQUENCE
            OBJECT IDENTIFIER	ecdsaWithSHA256
        SEQUENCE
            SET
                SEQUENCE
                    OBJECT IDENTIFIER	commonName
                    UTF8String	Trezor FIDO Root CA Serial 84151356
        SEQUENCE
            UTCTime	200402135903Z
            GeneralizedTime	20500402135903Z
        SEQUENCE
            SET
                SEQUENCE
                    OBJECT IDENTIFIER	countryName
                    PrintableString	CZ
            SET
                SEQUENCE
                    OBJECT IDENTIFIER	organizationName
                    UTF8String	SatoshiLabs, s.r.o.
            SET
                SEQUENCE
                    OBJECT IDENTIFIER	organizationalUnitName
                    UTF8String	Authenticator Attestation
            SET
                SEQUENCE
                    OBJECT IDENTIFIER	commonName
                    UTF8String	Trezor FIDO Serial 54878404
        SEQUENCE
            SEQUENCE
                OBJECT IDENTIFIER	ecPublicKey
                OBJECT IDENTIFIER	prime256v1
            BIT STRING	0004d918bdfa8a54ac92e90da91fca7aa26454c0d17336314dde83a54b86b5df4ef052659a1d6ffcb7467f1acddb8a33080b5eed918913f443a5261bc77b68606fc1
        cont 3
            SEQUENCE
                SEQUENCE
                    OBJECT IDENTIFIER	1.3.6.1.4.1.45724.1.1.4
                    OCTET STRING	0410d6d0bdc362eec4dbde8d7a656e4a4487
                SEQUENCE
                    OBJECT IDENTIFIER	basicConstraints
                    BOOLEAN	TRUE
                    OCTET STRING	3000
    SEQUENCE
        OBJECT IDENTIFIER	ecdsaWithSHA256
    BIT STRING	00304502202c04d0de06536d5ba6e87bebe06b2ea21d8eeb2b130e1bfdb7ed04368026d553022100cc393db0af3ad685c9bc359d345515b00d6368a0a6f013556362518541a01df6
```

onvej-sl, 2020-04-03:
I think "EE" stands for "end entity".

Are you sure that BasicConstraints.CA=false means that the certificate shouldn't be self-signed? I briefly went through the certificate validation process and it seems that the cA flag is required only for the non-leaf nodes in the certification path. On the other hand the standard states that a trust anchor (as input to the validation process) is an certification authority, which implies that an end certificate cannot be trust anchor. By the way, I tried issuing a self-signed certificate without the flag and verifying it (with itself as trust anchor) and I succeed.

andrewkozlik, 2020-04-03:
> I think "EE" stands for "end entity".
That's probably correct.

> Are you sure that BasicConstraints.CA=false means that the certificate shouldn't be self-signed? 

I am not sure and I haven't done much research into this, but the X.509 spec states:
> Use of self-issued certificates and self-signed certificates issued by other than certification authorities (CAs) are outside the scope of this Recommendation

Which indicates to me that the behavior is undefined and some RPs might not accept the certificate. So to maximize interoperability, it seems best to add a root.

onvej-sl, 2020-04-03:
> I am not sure and I haven't done much research into this, but the X.509 spec states:
>
>
> Which indicates to me that the behavior is undefined and some RPs might not accept the certificate. So to [...]

I didn't see that. I agree that it's probably undefined.

The funny thing is that it is not clear from the validation process description whether self-signed end point certificates should be rejected or not.

closed 2020-04-07 by andrewkozlik

tsusanka, 2020-04-23:
@andrewkozlik would like to write some specific FIDO/U2F test cases for QA or this needs just a general FIDO/U2F test?

andrewkozlik, 2020-04-23:
> @andrewkozlik would like to write some specific FIDO/U2F test cases for QA or this needs just a generic FIDO/U2F test?

Just the generic FIDO2/U2F tests with Firefox, Chrome and Edge. For FIDO2 this change should only affect registration at login.live.com and for U2F it affects all registrations.

sorooris, 2020-06-01:
Devices:
T1 1.9.1 revision 81553df0
TT 2.3.1 revision 5209804f

Tested:
webauthn.io - Register & Login
login.live.com - Register & Login
binance.com - Register & Login
dropbox.com - Register & Login
github.com - Setup & Login
google.com - Setup & Login
u2f.bin.coffee - Setup & Login

**QA OK**.

sorooris, 2020-06-02:
Tested:
webauthn.io - Register & Login
login.live.com - Register & Login
binance.com - Register & Login
dropbox.com - Register & Login
github.com - Setup & Login
google.com - Setup & Login
u2f.bin.coffee - Setup & Login

An issue appeared when testing Binance login; @wendys-cats will describe it below.

wendys-cats, 2020-06-02:
Binance bug reported as a separate issue [here](https://github.com/trezor/trezor-firmware/issues/1032).