trezor/trezor-firmware #933: passphrase entry on the device is persistent when passphrase is turned off/on

bosomt, 2020-04-06:
**Is your feature request related to a problem? Please describe.**

When you turn on passphrase entry on the device via 
trezorctl set passphrase enabled -f
Then there is no way to turn it of without trezorctl.

When you turn off and on passphrase via wallet then you are still asked to enter passphrase on device.

**Describe the solution you'd like**

Please consider that this option/feature is turned of when you disable passphrase via wallet then this feature is disabled too or when you enable passphrase via wallet you are asked if you want to enable this feature

prusnak, 2020-04-06:
`trezorctl set passphrase enabled -F` (capital F) is used to turn off the feature

tsusanka, 2020-04-08:
Let me rephrase. If I set "force on device" (`trezorctl set passphrase enabled -F`), then I disable the passphrase and then turn it on again, the "force on device" setting stays.  Shouldn't the "force" setting clear on passphrase disabling? cc @matejcik

prusnak, 2020-04-08:
>  Shouldn't the "force" setting clear on passphrase disable?

It probably should, but I think it's not implemented that way (in the firmware). If we are going to couple these two settings together (and I think we probably should, because it's less surprising), we should do it in the firmware.

matejcik, 2020-04-08:
Yes, if we want to change this, firmware should do it.

From a different angle, won't the Suite have the passphrase always on?

tsusanka, 2020-04-08:
> we should do it in the firmware.
I agree.
> From a different angle, won't the Suite have the passphrase always on?
That's true, as far as I know. This issue still makes sense for trezorctl or other hosts though (but it makes it less important, yes).

closed 2020-04-28 by tsusanka

sorooris, 2020-06-01:
QA OK. 

Device:
- TT 2.3.1 revision 5209804f

Tested against:
- MyTrezor [baking crocodile branch](https://mytrezor.corp.sldev.cz/baking-crocodile/development/#/) 97ec35
- Suite [fix/zcash-sign branch](https://suite.corp.sldev.cz/suite-web/develop/wallet/send/?trezor-connect-src=https://connect.corp.sldev.cz/fix/zcash-sign/) 30cb3eb