trezor/trezor-firmware #946: Run device tests on an actual TT in CI

tsusanka, 2020-04-14:
Related to #935. We need to add support to [tpmb](https://github.com/mmahut/tpmb) for TT first.

tsusanka, 2020-05-28:
On today's meeting we have realized we can do this also without tpmb (with the debug mode on of course).

We could modify the bootloader and:
- Always enter the bootloader, do not require any screen touch to enter.
- Do not overwrite the unofficial bootloader using firmware.
- Do not wait for the click when the firmware is loading up even if the firmware header says so.

cc @prusnak

hiviah, 2020-08-10:
Also the firmware should be immediately booted after loading it via bootloader and not reset (otherwise we would get bootloader bootloop).

hiviah, 2020-08-10:
The bootloader will delete storage on boot to make testing deterministic.

tsusanka, 2021-03-04:
Let's wait for the "workflow separation" task currently being done by @matejcik .

mmilata, 2021-03-22:
monero tests take about 10m to run on hardware, should we add them too?

matejcik, 2021-03-22:
yeah, let's

matejcik, 2021-04-22:
with #1565, the only tests that fail on hardware are the following:
```
FAILED ../../tests/device_tests/test_msg_signtx_bcash.py::TestMsgSigntxBch::test_send_bch_multisig_change
1198FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_bitcoin_gold_nochange
1199FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_p2sh
1200FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_attack_change_input
1201FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_multisig_1
1202FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_btg_multisig_change
1203FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_btg_external_presigned
1204FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_mixed_inputs
1205FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_p2sh_witness_change
1206FAILED ../../tests/device_tests/test_msg_signtx_bgold.py::TestMsgSigntxBitcoinGold::test_send_bitcoin_gold_change
```
the Bitcoin Gold failures are extra weird because even simple tests like `test_send_p2sh` fail. This is something to look into.

PR #1565 (2021-05-06): App separation and workflow restarts
should fix #1114

also fixes #946 

I recommend you review this not as a whole but going by individual commits. That way it's much more readable what is going on:

5fa959a27e38680a31fd556804b82f9e6b7a62f1 ensures that buffers in `trezor.wire` are allocated at import time, i.e., in a more predictable place

aa2504dba5ed27da9815a3bacd12b8e3e3e33c3f just moves `empty_bytearray` to `trezor.utils` where it actually fits much better

d6d4c0d9af39815f33f483645e2e0d5ca32e03f9 refactors `storage.cache` to preallocate all the space it needs at import time. It also mo [... cut]
closed 2021-05-06 by matejcik