trezor/trezor-firmware #948: Add Cardano Shelley support

MichalPetro, 2020-04-14:
**Is your feature request related to a problem? Please describe.**
Cardano cryptocurrency will switch to Shelley era in following months. Currenty Cardano Trezor model T integration has to be updated in order to allow the device to interact with the updated Cardano blockchain.

**Describe the solution you'd like**
Vacuumlabs team supported by Cardano Foundation and IOHK will implement required changes. We could use some contact person from Trezor that will help us with the process.

**Describe alternatives you've considered**
No alternatives

**Additional context**
The date of Cardano Shelley hard-fork is not known yet but it is expected in late Q2 2020 or Q3 2020.  Shelley is introducting new address formats, new structure of transactions, certificates that can be added to transactions and staking support. Currently we plan only to implement staking delegation features for HW wallets (so no support for stake pool operators) but this can change in the future. No need to add any new crypto primitives. We plan to add multisig support.

Delegation design specs (Shelley address structure (chapter 3.2), Shelley certificates (chapter 3.4)) 
https://hydra.iohk.io/build/2006688/download/1/delegation_design_spec.pdf

Formal spec
https://hydra.iohk.io/build/2363389/download/1/ledger-spec.pdf

Address serialization CDDL
https://github.com/input-output-hk/cardano-ledger-specs/blob/c9254440a2dd755856c523f1148adf50408639a5/shelley/chain-and-ledger/executable-spec/cddl-files/shelley.cddl

prusnak, 2020-04-15:
I'd suggest providing just a minimum required changeset (i.e. no staking, no staking delegation, no multisig) as a first pull request. We plan to have the next firmware release in 6 weeks and I can give no guarantee we'll have time to review more than the basic minimal change in that window.

MichalPetro, 2020-04-16:
@prusnak when do you expect to have following firmware release after that?

prusnak, 2020-04-16:
We do firmware releases on the first Wednesday of the month. But sometimes we skip the month.

Most probably it will be 8 weeks after that (making it 14 weeks in total).

prusnak, 2020-04-16:
@michalpetro I read that Shelley hard fork is switching from problematic Ed25519Bip32 to plain BIP44/BIP32 scheme. Can you please confirm or point me to a place where this is documented?  The Cardano website is a mess and I was unable to find anything. :-/

MichalPetro, 2020-04-17:
@prusnak The wallet scheme and crypto used in Shelley is fully described in this document: https://github.com/input-output-hk/cardano-wallet/wiki/Wallet-Cryptography-and-Encoding

It's a variation of BIP-44 using ed25519-bip32 and some other changes regarding the derivation path.

prusnak, 2020-04-17:
> https://github.com/input-output-hk/cardano-wallet/wiki/Wallet-Cryptography-and-Encoding
@matejcik the link above might be relevant for your Keychain rewrite - maybe we can get rid of the Cardano-specific part if we switch to the Byron scheme? (It would be a good time to do this breaking change now that Cardano is changing address format again ... sigh)

matejcik, 2020-04-17:
unfortunately a breaking change at this level would lose user keys, so i don't think we can do that :/

MichalPetro, 2020-04-23:
@prusnak we would like to hear your opinion on how to maintain backwards compatibility (which should be maintained for a while)

There are three entry points into the Trezor code related to Cardano. The protobuf messages CardanoGetAddress and CardanoGetPublicKey can be taken care of by adding an optional field for the protocol version. For CardanoSignTx, we can do the same, but there are fields that are no longer relevant in Shelley (transactions_count), some that have to be added, and it might be necessary to change some of the related types (CardanoTxInputType, CardanoTxOutputType), which would lead to a rather messy interface.

Another option is to create an entirely new message type for Cardano Shelley transaction signing.
Do you have some preferences in this matter, or perhaps some general guidelines for maintaining backwards compatibility?

matejcik, 2020-04-23:
What is the point of backwards compatibility here? IOW, why would you want to sign pre-Shelley transactions once the hard-fork is active?

MichalPetro, 2020-04-27:
@matejcik the main reason is to have the support for the Byron until the last day and to have support for Shelley since the first day. After the fork we can depreciate the Byron support.

prusnak, 2020-04-27:
This really complicates the code. I would vote for not having both chains supported in our code. Users can stick to older firmware when they need to transact on older chain and update their firmware only after the hardfork has happened.

mark-stopka, 2020-05-05:
@MichalPetro,

> so no support for stake pool operators

Would you be interested in collaborating on this? Is stake-pool operators support planned for Ledger?

mark-stopka, 2020-05-05:
> @matejcik the link above might be relevant for your Keychain rewrite - maybe we can get rid of the Cardano-specific part if we switch to the Byron scheme? (It would be a good time to do this [...]

Yeah, the old code would have to remain there for some time, so that users can migrate from old format to new format.

@prusnak, it may take users quite a while to update their Firmware (you know those HODLers :-) ) and if you remove the Byron support, they would not be able to migrate to the new Shelley addresses IMHO.

prusnak, 2020-05-05:
>  they would not be able to migrate to the new Shelley addresses 

They could do a downgrade

matejcik, 2020-05-06:
> Yeah, the old code would have to remain there for some time, so that users can migrate from old format to new format.
>
> @prusnak, it may take users quite a while to update their Firmware (you know [...]

please explain:
Does Shelley make entirely new addresses, with new keys?
Would I need to "migrate" = send my funds from a Byron address to a Shelley address?
Once the hardfork becomes active, would I need Shelley-compatible Trezor with Byron key derivation to spend my pre-Shelley funds, i.e., to migrate?

The above-linked page has no mention of the "scheme" being used with Shamir backup - which is what we would ideally switch to in all cases, but that would lose us recoveryseed compatibility with Cardano spec -- unless Shelley also does modifications in this regard?

MichalPetro, 2020-05-06:
> @MichalPetro,
> 
> > so no support for stake pool operators
> 
> Would you be interested in collaborating on this? Is stake-pool operators support planned for Ledger?

After discussion with IOHK it was decided that this will be not implemented on Ledger or Trezor but this may be included in the future.

MichalPetro, 2020-05-06:
@matejcik 

> Does Shelley make entirely new addresses, with new keys?

Shelley has new addresses but legacy addresses will still be supported (there will be several addresses formats). You can access your Byron-funds with Byron-key after Shelley update

> Would I need to "migrate" = send my funds from a Byron address to a Shelley address?
No

> Once the hardfork becomes active, would I need Shelley-compatible Trezor with Byron key derivation to spend my pre-Shelley funds, i.e., to migrate?

The above answers answer also this question. Yes you will need Shelley compatible Trezor with Byron (or Shelley) keys but no need to migrate.

matejcik, 2020-05-07:
if that's the case, users won't actually be "migrating" away from Byron keys, so we will need to keep that support.

That's just the key data though. All the other code can be switched to Shelley, IIUC

MichalPetro, 2020-06-03:
@matejcik does it make sense to submit some WIP PR to you for a preliminary review or should we only submit once we are done?

mark-stopka, 2020-06-03:
@MichalPetro I would like to have a look if you guys don't mind...

MichalPetro, 2020-06-03:
@mark-stopka here is the public repo that we work in.  https://github.com/vacuumlabs/trezor-firmware

If Trezor confirms that they want a sneak peak on the WIP, we will clean it up a bit before submitting.

matejcik, 2020-06-03:
Please submit a draft PR. No need to clean up, we won't be reviewing it right now anyway, but having a sneak peek at the diff would be nice.

gabrielKerekes, 2020-06-05:
Hi guys, so I've submitted the draft PR with our progress --> #1048. Feel free to check it out and let us know if you need anything.

gabrielKerekes, 2020-06-09:
Trezor Connect library will also need to be updated to work with the Trezor Shelley update. We are not sure how we should go about this. Do we update it? If so, do we smiply create a PR there or how does it work?

Also I am having some trouble building the trezor firmware for an actual device - I've tried it only once so far and I will try it once more, but maybe someone will have some quick pointers. I've tried the steps mentioned [here](https://docs.trezor.io/trezor-firmware/core/build/embedded.html) and also the docker build from [here] (https://wiki.trezor.io/Developers_guide:Custom_firmware). The `make` steps fail (even on current master without our changes), because of a missing `NULL` definition and the docker build fails, because of an invalid signature? Do you have any idea what I might be doing wrong? Or maybe is there someone who I could get in touch with, if I'm not able to get this to work? 

Thanks!!

matejcik, 2020-06-09:
> Trezor Connect library will also need to be updated to work with the Trezor Shelley update. We are not sure how we should go about this. Do we update it? If so, do we smiply create a PR there

yes

> The `make` steps fail (even on current master without our changes), because of a missing `NULL` definition

sounds like a toolchain version problem?

> and the docker build fails, because of an invalid signature?

try running `PRODUCTION=0 ./build-docker.sh`

matejcik, 2020-06-22:
while you folks are on this, care to look at #438 as well? thanks!

MichalPetro, 2020-07-06:
We prepared the first PR here https://github.com/trezor/trezor-firmware/pull/1096
The Second PR will follow shortly this week.
First 2 PR's are crucial for sending funds
The third PR will introduce staking support.

MichalPetro, 2020-07-09:
@matejcik any idea who will be reviewing our PRs?

gabrielKerekes, 2020-07-17:
The second PR can be found here #1112. We still hope there's a chance for even the third PR to make into the release. It shouldn't contain too much new code to review. It should be ready early next week.

gabrielKerekes, 2020-07-21:
Is the July 24 deadline also valid for Trezor Connect? Ideally we would like to be able to submit the Trezor Connect PRs until next Tuesday - July 28. Would that be possible?

tsusanka, 2020-07-21:
It is not, they have different release process. On the other hand they are really under-staffed at the moment so we can't make any promises when they will be able to review it.

gabrielKerekes, 2020-07-21:
Then what will happen if new Trezor FW is released (with Shelley updates) and Trezor Connect with Shelley support has not yet been released?

tsusanka, 2020-07-21:
We will do our best to make it happen. Are the changes in Connect big?

gabrielKerekes, 2020-07-21:
That depends on what big means. The second PR will be slightly larger than the [first PR](https://github.com/trezor/connect/pull/623) because of larger protobuf changes, but I don't think it should be too bad. There should only be changes in the types and their validation.

tsusanka, 2020-07-24:
@gabrielKerekes is the date for the Shelley hard-fork still July 29th?

tsusanka, 2020-07-24:
Closed by #1096 and #1112 and waits for PR number 3.

closed 2020-07-24 by tsusanka