trezor/trezor-firmware #949: Soft-locking for Trezor T

matejcik, 2020-04-14:
This is a pre-requisite for #75. In order to implement auto-locking, we first need to have **locking**.

There must be a "soft-locked" mode, in which the user needs to enter PIN before proceeding. However, exact details of the soft-locked mode are unclear.

We are assuming that:
* FIDO should remain active in soft-locked mode, i.e., authentication attempts shouldn't be blocked by PIN
* passphrases (and presumably sessions) should stay cached
* USB should stay on

That means that we cannot lock storage when soft-locked. We also need to respond to USB messages somehow.

Per @slush0 in #75, "wake event should be a human interaction (touch event)".

I proposed three distinct behaviors for the soft-lock:

1. Identical to T1: device responds on USB normally, but asks for PIN whenever the user tries to do something with seed. The "ask for PIN" is inline, so:
   ```
   >> GetAddress
   << ButtonRequest(Enter PIN)
   >> ButtonAck
   (user enters PIN)
   << Address
   ```
   There are some implementation obstacles to this, so it's not preferred from my side, but it is solvable.
2. Everything needs PIN: same behavior as above, except PIN is requested for _every_ message. User-unfriendly, because even things like `Initialize` would prompt; but could be solved in cooperation with Suite and some special-casing
3. Respond to *every* USB message with `Failure(Device is locked)` (incl. `Initialize` and other stuff). Wait for user to tap the screen to unlock.
   This would be my preference from the implementation side, it is the "most correct" way to be soft-locked, and it matches the "wake event should be human interaction" best.
   The drawback is that it introduces a completely new mode of operation that the Suite (and other clients) would need to learn about.

This needs input from Product. Please confirm or reject our assumptions and give us some opinions about the mode of operation.

matejcik, 2020-04-16:
after conferring with Product, we settled on the following:

* USB stays on, for several reasons
* Storage _could_ be locked, but that is an implementation detail.
* Passphrases should remain in cache (which makes storage lock kind of pointless).
* FIDO should be locked.
* Device should respond to `Initialize` and `GetFeatures` normally, and require PIN for all other operations. PIN prompt should be inline, as in (1), and software will cause the PIN dialog to show
* Device should _also_ respond to user touch by prompting for PIN (useful for FIDO)
* Locked state should be indicated on the device display, ideally with lowered brightness
* Locked state should be indicated in Features message, presumably with `pin_cached` field?

one remaining question: what happens when there is a FIDO request?
Should the device ask for PIN directly?
Should the request be rejected until the user unlocks?

prusnak, 2020-04-16:
> Locked state should be indicated on the device display, ideally with lowered brightness

We already have the lock screen (home screen + with the text LOCKED on top) visual. Let's use this one (not the code as it happens in `boot.py`, visual only).

>  Should the device ask for PIN directly?

This would probably be the best UX.

>  Should the request be rejected until the user unlocks?

If we for some reason decide not to ask for the PIN directly, the request should be ignored, not rejected (see https://github.com/trezor/trezor-firmware/pull/947 for reason).

matejcik, 2020-04-16:
> > Locked state should be indicated on the device display, ideally with lowered brightness
> 
> We already have the lock screen (home screen + with the text LOCKED on top) visual. Let's use this one (not the code as it happens in `boot.py`, visual only).

I've been using the same graphic in experiments, yes. But I think there should be a visual distinction; more for the sake of the start-up screen though. It would be nice if the device could indicate that it is invisible to host.

prusnak, 2020-04-16:
>  It would be nice if the device could indicate that it is invisible to the host.

Good point. Still, the screens should be IMHO somehow similar. 

[image]

tsusanka, 2020-04-16:
@matejcik summarized it very well, although it is probebly clear I want to add this terminology summary:

|   | soft lock | hard lock |
|---|---|---|
| USB  | YES  | NO |
| FIDO/U2F  | NO  | NO  |
| Initialize  |  YES  | NO |
| other messages | NO | NO |
| storage | _could be_ | locked |
| passphrase cache | intact | empty |
| visual | home screen lock | a bit different home screen lock |
| on touch | entry PIN dialog | entry PIN dialog |

prusnak, 2020-04-16:
I think it should be

|   | soft lock | hard lock |
|---|---|---|
| FIDO/U2F  | YES | NO  |

in the table above if we decide the device should ask for PIN directly on FIDO request during the soft lock.

tsusanka, 2020-04-16:
Hmm, so maybe like this?

|   | soft lock | hard lock |
|---|---|---|
| FIDO/U2F  | entry PIN dialog | NO  |

tsusanka, 2020-04-16:
On the other hand that is what we will do also if "other messages" come, isn't it?

andrewkozlik, 2020-04-17:
On Trezor 1, U2F does not require PIN entry in locked state (neither soft nor hard). In FIDO2 the client tells us whether we should request PIN before authentication.

andrewkozlik, 2020-04-17:
On Trezor 1, soft-lock locks the storage and we use the state of the storage to determine whether the device is locked or not, i.e. device lock is equivalent to storage lock.

matejcik, 2020-04-17:
> On Trezor 1, U2F does not require PIN entry in locked state (neither soft nor hard). In FIDO2 the client tells us whether we should request PIN before authentication.

in discussion with @slush0 we agreed that we want to require PIN in soft-lock even if the other party doesn't indicate that; Trezor soft-lock is supposed to protect you when you step away from your PC, and allowing PIN-less 2FA goes against that.

what is the state of PIN support in FIDO? what do the flows look like?

i'm thinking, if a FIDO request arrives on a softlocked device, should the PIN dialog show first, or should "authorize? yes/no" show first?
maybe the "OK" button should be replaced by green "Enter PIN" in that situation?

matejcik, 2020-04-17:
> On Trezor 1, soft-lock locks the storage and we use the state of the storage to determine whether the device is locked or not, i.e. device lock is equivalent to storage lock.

we might be able to do the same thing on TT, but isn't it sort of pointless when there are cached seeds in RAM? (true on T1 as well)

andrewkozlik, 2020-04-17:
Just to be clear, in the above I am not saying either is right or wrong, simply stating the facts, because one of the requirement was to match T1 behavior. I also want to emphasize, that there clearly was a strong requirement to be able to use U2F in hard-locked state on T1, because it required a specific hack in the implementation. This obviously has to do with the way the PIN is entered on T1.

FIDO: Currently we first display the login request. If the user approves, then we check whether the relying party requested user verification. If so, then we ask for the PIN before signing the authentication response.
- What I like about this is that you know why you are being asked for the PIN and you can also decline the request before entering the PIN.
- A possible disadvantage is that it may reveal to the holder of the device, whether the credential belongs to the device or not, but I think that's acceptable.
- If we decide to lock the storage, then in case of passwordless logins we would first need to ask for the PIN in order to access resident credentials in the storage.

Storage: I prefer to err on the side of caution, so I would rather ask: Is there reason not to lock the storage? Firstly not all data from the storage is cached in RAM, but I agree that indeed the most valuable thing, the seed/nodes, are cached. Secondly I think the cached nodes should be encrypted in RAM when the device is locked.

tsusanka, 2020-04-17:
I have an action item to summarize how the locks work on T1 and TT and then talk to Product to document and specify it properly. For example, the requirement "to match T1 behavior" seems to be lifted as of now (as discussed with Product). In other words, this needs a document to make a clear specification.

matejcik, 2020-06-04:
fixed via #958

closed 2020-06-04 by matejcik