trezor/trezor-firmware #957: Encrypt sessions when device is soft-locked

matejcik, 2020-04-22:
Soft-locked devices lock the storage, so that PIN is required to access private data.

However, derived seeds with passphrase applied are kept in RAM, so the soft-lock doesn't offer meaningful protection.

Secrets should also be encrypted in RAM when the device is locked.

tsusanka, 2020-04-23:
We should also revisit that cache.py is the only place where the caching occurs (not in trezor-crypto etc.). For example that we do not cache nodes anywhere else.

fresheneesz, 2021-01-28:
I came here to say this. Wouldn't this solve the [infamous trezor vulnerability](https://www.ledger.com/improving-the-ecosystem-disclosure-of-the-trezor-recovery-phrase-extraction-vulnerability)?