trezor/trezor-firmware #960: Update token list - RSK Chain - RIF

jonathansmirnoff, 2020-04-22:
I have noticed that it is referenced to an older version of ethereum token list (https://github.com/ethereum-lists/tokens/tree/88414ef852e21279f217d2f3f8fd439cd7c5fd9f) in https://github.com/trezor/trezor-firmware/tree/master/common/defs/ethereum. 

Recently the ethereum-lists project has been updated and I was wondering when you will update the reference. Also this question it is related because it has been added a new token: RIF. This a token from RSK chain. My goal is to allow Trezor to decode the erc20 token transfer transaction in RSK chain.

jonathansmirnoff, 2020-04-23:
Also I'm ready to help with this.

matejcik, 2020-05-21:
fixed in eeb46eabca24f40e60d41be31b44703d4964e578

closed 2020-05-21 by matejcik