trezor/trezor-firmware #961: U2F authentication broken in 1.9.0 in Github

bosomt, 2020-04-23:
**Describe the bug**
When you login to GitHub device asks you to Register U2F device
Not asking you to authenticate but to register

**Firmware version or revision**
1.9.0 broken
1.8.3 works fine

**Desktop/smartphone setup (please complete the following information):**
nixos
firefox 75.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'github.com'
2. Login via security key
3. use firmware 1.9.0
4. See error

**Expected behavior**
1.8.3 firmware says Authenticate

**Screenshots**

[image]

prusnak, 2020-04-23:
key 000000 means that you have another token registered with the service

bosomt, 2020-04-23:
@prusnak yes i have couple of them registered

+ when you try to add it again you won't get usual response that it's already registered but this:

[image]

bosomt, 2020-04-23:
but this error is same with 1.8.3 FW

andrewkozlik, 2020-04-23:
I think I know exactly what this is about. FF uses a different BOGUS_APPID than Chrome. We only check the Chrome BOGUS_APPID in T1 code. Trezor T should be handling this correctly.

@bosomt Please also check Edge and Safari browsers.

bosomt, 2020-04-23:
@andrewkozlik Win10+Edge and MAC+Safari works fine

prusnak, 2020-04-23:
So maybe we should use the 000000 bogus ID to T1 codebase same way we did for TT.

andrewkozlik, 2020-04-23:
@bosomt I prepared a fix here: https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/523731139/artifacts/download.

There are two error scenarios:
1. Trying to register a token which is already registered.
2. Trying to log in with a token which isn't registered.

In Firefox both should work correctly now. In Chrome, the first one doesn't work nicely, it shows an authentication request. I think I investigated this a while back and found that we can't do anything about it, it's a Chrome issue. Can you elaborate on what doesn't work with Edge and Safari? Scenario + what appears in the browser + what appears on Trezor.

bosomt, 2020-04-24:
FF: says not registered when not registered
FF: when registered , works fine

EDGE: when not registered: unknown key and Trezor has no message, just homepage
EDGE: when registered , works fine

Safari: im not at office anymore, so im not able to test it,

**FIXED**

bosomt, 2020-04-24:
sorry I forgot about that 1. scenario.

this one is broken

1) you add token, this works fine
2) add one more time, this will ask you: **Authenticate U2F security key** ? 
3) press Authenticate : Already registered

so it works properly but it suppose to ask add Register not Authenticate,  but we can say that it indicates that it's already registered ;)

andrewkozlik, 2020-04-24:
> so it works properly but it suppose to ask add Register not Authenticate, but we can say that it indicates that it's already registered ;)

Unfortunately, I think we can't do anything about that on our side. As long as the user gets the information about the token being already registered, it's OK.

bosomt, 2020-04-24:
@andrewkozlik great, so closing

closed 2020-04-24 by bosomt

danielg4, 2025-01-11:
Current firmware here. Just tried to add a Passkey in GitHub a short time ago, and T1 said it was already registered on another device, meaning Chrome was sending its bogus AppId. This is just speculation on my part, but maybe GitHub is trying FIDO2 first, and falling back to FIDO1, which the T1 actually supports, but the legacy firmware gets confused by the way Chrome handles the fallback?