trezor/trezor-firmware #964: `imported` field behavior in Features

tsusanka, 2020-04-24:
@trezor/qa please test:

- load T1 1.8.0 using `trezorctl device load -s`
- check get-features (`imported` field should be present)
- do firmware upgrade to latest
- check get-features again

bosomt, 2020-04-24:
@tsusanka here:
1.8.0

trezorctl get-features
Features (134 bytes) {
    bootloader_hash: 32 bytes 0xf7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8,
    device_id: '614DF938FF571B9D4D435F13',
    imported: True,
    initialized: True,
    label: 'SLIP-0014',
    major_version: 1,
    minor_version: 8,
    model: '1',
    needs_backup: False,
    no_backup: False,
    passphrase_protection: False,
    patch_version: 0,
    pin_cached: False,
    pin_protection: False,
    revision: 20 bytes 0x964a622bb512aa85cfcc3e451fc70729cc15bb4f,
    unfinished_backup: False,
    vendor: 'trezor.io',

1.9.0

trezorctl get-features
Features (203 bytes) {
    bootloader_hash: 32 bytes 0xf7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8,
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Crypto (5),
        Ethereum (7),
        Lisk (8),
        NEM (10),
        Stellar (12),
        U2F (14),
    ],
    device_id: '614DF938FF571B9D4D435F13',
    imported: True,
    initialized: True,
    label: 'SLIP-0014',
    language: 'en-US',
    major_version: 1,
    minor_version: 9,
    model: '1',
    needs_backup: False,
    no_backup: False,
    passphrase_protection: False,
    patch_version: 0,
    pin_cached: False,
    pin_protection: False,
    revision: 20 bytes 0x0b7a8449f8dd003fc415262b05102d113247d3de,
    session_id: 32 bytes 0x248e450ed9e63a94f953aa0823752f28825ce9011b024caac2359bb78fd6a885,
    unfinished_backup: False,
    vendor: 'trezor.io',
    wipe_code_protection: False,

anything else i can do ?

tsusanka, 2020-04-24:
https://github.com/trezor/trezor-firmware/issues/964

matejcik, 2020-04-24:
i'm not sure if we care about the `imported` value in itself? maybe it should be removed from protobuf?

prusnak, 2020-04-24:
We do care about this. For example, we can decided to show a warning in Suite that the seed was not generated in the device, but somewhere else (not securely) and only imported into the device.

matejcik, 2020-04-24:
in that case, the behavior is correct. closing.

closed 2020-04-24 by matejcik