trezor/trezor-firmware #965: XPRV in latest legacy firmware

tsusanka, 2020-04-24:
@trezor/qa please test:

- load T1 with xprv, you need firmware 1.8.3 or lower and probably older trezorctl
- check address
- upgrade to latest
- check address again (this will fail most likely)

We would like to know _how_ it fails.

bosomt, 2020-04-24:
@tsusanka 
beta wallet says: Loading of accounts failed + ["log","Fri, 24 Apr 2020 13:14:06 GMT","[trezor.js] [call] Received","Failure",{"code":"Failure_NotInitialized","message":"Device not initialized or passphrase request cancelled or unsupported curve"}]

old wallet says: Reading of accounts failed.
Error details: Error: The coin Bitcoin is not supported by your Trezor.

bosomt, 2020-04-24:
loaded this trezorctl load-device -x xprv9s21ZrQH143K24Mfq5zL5MhWK9hUhhGbd45hLXo2Pq2oqzMMo63oStZzF93Y5wvzdUayhgkkFoicQZcP3y52uPPxFnfoLZB21Teqt1VvEHx

+ 1.8.3
what do you recommend ? older FW ? older wallet ?

matejcik, 2020-04-24:
> @tsusanka
> beta wallet says: Loading of accounts failed + ["log","Fri, 24 Apr 2020 13:14:06 GMT","[trezor.js] [call] Received","Failure",{"code":"Failure_NotInitialized","message":"Device not initialized or passphrase request cancelled or unsupported curve"}]

is this on 1.8.3 or after upgrade?

prusnak, 2020-04-24:
How does the Feature message look?

bosomt, 2020-04-24:
trezorctl get-features
WebUSB transport is disabled: libusb-1.0.so: cannot open shared object file: No such file or directory
Features (159 bytes) {
    bootloader_hash: 32 bytes 0xf7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8,
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Crypto (5),
        Ethereum (7),
        Lisk (8),
        NEM (10),
        Stellar (12),
        U2F (14),
    ],
    device_id: '9FD6BFDEACA691184AEFA0F1',
    imported: True,
    initialized: True,
    language: 'english',
    major_version: 1,
    minor_version: 8,
    model: '1',
    needs_backup: False,
    no_backup: False,
    passphrase_cached: False,
    passphrase_protection: False,
    patch_version: 3,
    pin_cached: False,
    pin_protection: False,
    revision: 20 bytes 0xdf0963ec48f01f3d07ffca556e21ff0070cab099,
    unfinished_backup: False,
    vendor: 'trezor.io',
}

@matejcik before upgrade

bosomt, 2020-04-24:
after update to 1.9.0

trezorctl get-features
WebUSB transport is disabled: libusb-1.0.so: cannot open shared object file: No such file or directory
Features (156 bytes) {
    bootloader_hash: 32 bytes 0xf7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8,
    capabilities: [
        Bitcoin (1),
        Bitcoin_like (2),
        Crypto (5),
        Ethereum (7),
        Lisk (8),
        NEM (10),
        Stellar (12),
        U2F (14),
    ],
    device_id: '9FD6BFDEACA691184AEFA0F1',
    imported: True,
    initialized: True,
    language: 'english',
    major_version: 1,
    minor_version: 9,
    model: '1',
    needs_backup: False,
    no_backup: False,
    passphrase_protection: False,
    patch_version: 0,
    pin_cached: False,
    pin_protection: False,
    revision: 20 bytes 0x0b7a8449f8dd003fc415262b05102d113247d3de,
    unfinished_backup: False,
    vendor: 'trezor.io',

wallet/suite still not working with this PK :)

tsusanka, 2020-04-24:
So to summarize, after upgrading to 1.9.0 `Features.imported is True` and also `Features.initalized is True`, but at the same time getting an address fails on Wallet/Suite. What does the error say in Wallet/Suite?

prusnak, 2020-04-24:
This is a firmware bug (or rather inconsistency). If Trezor cannot perform key derivation (and it can't because the root node is not accessible) it should not claim it is initialized in the Features.

In another word, `config_isInitialized` behavior does not match `config_getSeed` behavior. I suggest adding `config_hasMnemonic` call function and 

Replacing this

https://github.com/trezor/trezor-firmware/blob/f054bf6e95b8089871e61b0db58794de480243d4/legacy/firmware/fsm_msg_common.h#L49

with the following:

```
resp->initialized = config_isInitialized() && config_hasMnemonic();
```

tsusanka, 2020-04-28:
@prusnak there is no `config_hasMnemonic`. We can either:
1. Introduce `hasMnemonic` and use `config_getMnemonic` and throw away (memzero) the result.
2. Introduce `hasMnemonic` and introduce "has item" in storage. So something like `storage_has(KEY_MNEMONIC)` would be used.
3. Replace `config_hasMnemonic` with `!config_getImported()`. So it would be `resp->initialized = config_isInitialized() && !config_getImported();`. 

(3) seems by far the easiest.

prusnak, 2020-04-28:
(3) does not seem right, does it? If the seed is imported, initialized will be false.

tsusanka, 2020-04-28:
I think I misunderstood what `imported` stands for. I thought it was equal to "was XPRV imported", but now I see it is also true for `trezorctl load-device [mnemonic]`. I will have another look and probably end up with (2), if it goes smoothly.

prusnak, 2020-04-28:
Correct. (2) seems to be the way.

tsusanka, 2020-04-28:
There is a another problem with this approach. Mnemonic is a confidential item but `intialized` bool is public. Therefore with this change `Features.initialized` becomes False when the device is locked (and it was True).

However, @matejcik discovered very similar issue while working on #949. We came to the conclusion we will not send the `intialized` flag when locked for TT and we are communicating this change to Suite. Tl;dr we could do the same here for T1.

tsusanka, 2020-05-19:
As this is dependent on #949 I am moving this to 2020-07 as well.

tsusanka, 2020-06-18:
Due to #1072 we have decided to do this a bit differently and we will set the Initialized field in storage to False if no mnemonic is present.

tsusanka, 2020-06-19:
@bosomt could you please retest with current master (4239e8d)?

- load T1 with xprv, you need firmware 1.8.3 or lower and probably older trezorctl
- do upgrade to master
- check that initialized in get-features is false

closed 2020-06-19 by tsusanka

bosomt, 2020-06-23:
@tsusanka 

I was able to start old trezorctl and perform test.
see: https://pastebin.com/Jyu9kf2t

Result: 

initialized in get-features is false after upgrade 

1.8.3 unsigned 6e13d2fa :     initialized: True,
1.9.2. current master 6ebb3889:     initialized: False,

PR #1075 (2020-06-19): Set initialized in storage to false if no mnemonic is present
Closes #965