trezor/trezor-firmware #968: Caching the passphrase (25th word) for a certain amount of time

a20eac1d, 2020-04-26:
**Is your feature request related to a problem? Please describe.**
When using the TREZOR I get asked to enter my passphrase (or "25th word") over and over again, even if I JUST entered it. Switching from one currency to another is kind of annoying but understandable, but Monero is especially bad with this.

For Monero you have to use they Monero Client. Even just switching tabs inside the Monero software (going from "Accounts" to "Receive") makes TREZOR request the passphrase again and again. It is the same with refreshing your balance. Every time Monero asks for a balance update, it requests the key from the TREZOR again, which in turn asks for the passphrase. With a long passphrase this is incredibly annoying and wastes a lot of time for a negligible security benefit.

The security implications of having the passphrase cached (in an encrypted way) are low, considering the fact that having to enter your password again and again increases the chances of somebody spying on you and reading the password you are entering from the TREZORs keyboard.

Caching could open the TREZOR for more attack vectors like reading the encrypted passphrase from memory while it is running, but an option to disable or limit the caching to (30, 60, 120, 300) seconds would give the user the option to choose depending on how paranoid they want to be.

**Describe the solution you'd like**
Give us the options to have TREZOR cache the passphrase for (30, 60, 120, 300) seconds. If the TREZOR has been inactive for longer, delete the cached passphrase and request it again from the user.

**Describe alternatives you've considered**
There are no alternatives that I can think of. This is, IMHO, an incredibly important feature for many users (including myself). Not using a passphrase is a bad solution and puts the users security at risk, but having to enter the passphrase over and over again is a horrible user experience (as anybody using Monero will confirm).

I am getting incredibly annoyed with TREZOR and Monero and the way they work together. It is really annoying to interact with both and having to enter my passphrase 10+ times every time I want to send or receive funds.

**Additional context**
-

prusnak, 2020-04-26:
This is not a Trezor's fault and will be fixed when https://github.com/monero-project/monero-gui/pull/2829 is merged.

If you are annoyed about the Monero UX you should really tell them, maybe even donate, they are an open-source community project of volunteers doing that in their spare time.

closed 2020-04-26 by prusnak

selsta, 2020-04-27:
> This will be fixed when monero-project/monero#6446 and monero-project/monero-gui#2829 are merged.

Both will be included in v0.15.1.0 which should be out really soon. (1-2 weeks).