trezor/trezor-firmware #976: Unable to sign custom firmware

Dimfred, 2020-05-03:
Hello, I am currently trying to build my custom firmware, but the `build_firmware` target fails on signing the firmware.

I run:
`BUILD_VERBOSE=1 PRODUCTION=0 BITCOIN_ONLY=0 PYOPT=0 make vendor build_firmware`

### Initial build output before signing

cat build/firmware/firmware.bin.p1 build/firmware/firmware.bin.p2 > build/firmware/firmware.bin
tools/binctl build/firmware/firmware.bin -h
Trezor Vendor Header
  * magic   : TRZV
  * hdrlen  : 2560
  * expiry  : 0
  * version : 0.0
  * scheme  : 2 out of 3
  * trust   : 65422 = [0 . . . 4 5 6 . . . . . . . . .] = [WAIT_1 RED CLICK STRING]
  * vpub 1 : e28a8970753332bd72fef413e6b0b2ef1b4aadda7aa2c141f233712a6876b351
  * vpub 2 : d4eec1869fb1b8a4e817516ad5a931557cb56805c3eb16e8f3a803d647df7869
  * vpub 3 : 772c8a442b7db06e166cfbc1ccbcbcde6f3eba76a4e98ef3ffc519502237d6ef
  * vstr    : UNSAFE, DO NOT USE!
  * vhash   : c5b4d40cb76911392122c8d1c277937e49c69b2aaf818001ec5c7663fcce258f
  * vimg    : (2347 bytes)
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : 3c596a48c56d356160aa543c753f24189a563638d6a3162c29edb7dfa52779daa3ea430c8f7670173425e38fff19c20bf6ef5b6e7989d003bf02366d65265208
  * fngprnt : 14304230ba8d25ddf539d6d435ca17ece5e3bd28fa87c678ff8e76c1b925bebe

No firmware signature
Trezor Firmware Image
  * magic   : TRZF
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 1549824
  * version : 2.2.1.0
  * fixver  : 2.2.0.0
  * hashes: OK
    - 00 : 1aaf8094aabcf69a06a458988f6c0bcf2c3b9b9e70c113e94a848f191d4829ef
    - 01 : b675c4c232155950f9e46715ffddbe000187d340a23bd97820eab03bb8277506
    - 02 : 52adf605c525ce11428482222c30662b5ede5ea32520a1c52ab4acfad61bb729
    - 03 : 4ce0341c740f4ee15149fd42fd6d03c1f06d82d9e29cf46f7b3a487a9aec540d
    - 04 : 403796227cc128522de2c780640ddddc74f7b0ce3197eac85db9fc69b1c11bcd
    - 05 : 22f04b226160fc0b8b6988fca203317190ccabaed5f311087faa98e283d62de7
    - 06 : 06c3a436e417bfeb282b193572c1b97edb3de8abdf5969755091298921564f73
    - 07 : 7f3671eef84930e876b2607a885525cce84a92e12d427f96a29ae5e0cbec9658
    - 08 : df417660217f4bc4f46a262828910ba62edeff53b63d23a8bd3f136cae7d8c7f
    - 09 : c2eea73b505823fc9d0390961de7d0d3290bdecf8e1cf1c048eda6435df9e475
    - 10 : b4cbb26f13c7a9d27a46686e173d2ecd5bd07e976f34c38a35a89eb1e1fec6da
    - 11 : 6a542e2e3218a8d2f42f4f700424996dc0b075f2b9644ed5d9871c3e70bcc6aa
    - 12 : 0000000000000000000000000000000000000000000000000000000000000000
    - 13 : 0000000000000000000000000000000000000000000000000000000000000000
    - 14 : 0000000000000000000000000000000000000000000000000000000000000000
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x00 = [. . . . . . . .]
  * sig     : 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  * total   : 1553408 bytes
  * fngprnt : a3adf98f6b54c79452edbd994e489a1b98eedde0c87c5bf4cbc85b82d0485fbc

### Signing process

tools/binctl build/firmware/firmware.bin -s 1:2 tools/keyctl sign firmware build/firmware/firmware.bin 4747474747474747474747474747474747474747474747474747474747474747 4848484848484848484848484848484848484848484848484848484848484848`
Traceback (most recent call last):
  File "tools/keyctl", line 73, in <module>
    cli()
  File "/home/dimfred/01_data/02_programs/pyenv/versions/3.7.3/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/dimfred/01_data/02_programs/pyenv/versions/3.7.3/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/dimfred/01_data/02_programs/pyenv/versions/3.7.3/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/dimfred/01_data/02_programs/pyenv/versions/3.7.3/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/dimfred/01_data/02_programs/pyenv/versions/3.7.3/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "tools/keyctl", line 68, in sign
    cosi.verify(sig, digest, global_pk)
TypeError: verify() missing 2 required positional arguments: 'keys' and 'mask'
Trezor Vendor Header
  * magic   : TRZV
  * hdrlen  : 2560
  * expiry  : 0
  * version : 0.0
  * scheme  : 2 out of 3
  * trust   : 65422 = [0 . . . 4 5 6 . . . . . . . . .] = [WAIT_1 RED CLICK STRING]
  * vpub #1 : e28a8970753332bd72fef413e6b0b2ef1b4aadda7aa2c141f233712a6876b351
  * vpub #2 : d4eec1869fb1b8a4e817516ad5a931557cb56805c3eb16e8f3a803d647df7869
  * vpub #3 : 772c8a442b7db06e166cfbc1ccbcbcde6f3eba76a4e98ef3ffc519502237d6ef
  * vstr    : UNSAFE, DO NOT USE!
  * vhash   : c5b4d40cb76911392122c8d1c277937e49c69b2aaf818001ec5c7663fcce258f
  * vimg    : (2347 bytes)
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : 3c596a48c56d356160aa543c753f24189a563638d6a3162c29edb7dfa52779daa3ea430c8f7670173425e38fff19c20bf6ef5b6e7989d003bf02366d65265208
  * fngprnt : 14304230ba8d25ddf539d6d435ca17ece5e3bd28fa87c678ff8e76c1b925bebe

No firmware signature
Traceback (most recent call last):
  File "tools/binctl", line 359, in <module>
    main()
  File "tools/binctl", line 349, in main
    signature = binascii.unhexlify(sys.argv[4])
IndexError: list index out of range
scons: *** [build/firmware/firmware.bin] Error 1
make: *** [Makefile:124: build_firmware] Error 2

### I am running with
    - Ubuntu 20.04
    - python 3.7.3
    - pipenv sync // all passed
    - all dependencies installed
    - not in docker

The `build_unix` target works perfectly and I could verify that everything works as expected with the emulator.

### how I build with docker

The docker build skips my probuf messages. (probably because, they aren't present in the official messages repo?)

`PRODUCTION=0 BITCOIN_ONLY=0 REPOSITORY=local TAG=<my_branch> ./build-docker.sh`

Any help appreciated, thanks in advance.

matejcik, 2020-05-04:
There is something seriously wrong with your checkout. It looks like you are using an old version of binctl/keyctl (which doesn't exist in the repository anymore) together with a relatively new trezorlib -- which should not exist at the same time as the above tools, unless you just picked a commit at random in the middle of the conversion.

Dimfred, 2020-05-04:
I created my branch from `trezor-firmware/master`. Seemed for me like the latest branch. If it isn't the right branch, which would it then be?

prusnak, 2020-05-04:
try the following - which brings the master branch to the latest commit:

```
git checkout master
git pull
git checkout <yourbranch>
git rebase master
```

Dimfred, 2020-05-04:
Okay awesome it worked! Thank you. I actually checked the online diff from my master against yours and they both seemed to be on the same commit. Anyways it worked.

I have fixed some issues related to the rebase and could verify everthing on the emulator.

Updating the firmware works now. But the trezor gets stuck in the starting up view. Everything loads, it says `done` but nothing else happens. Is there a way to debug this with having a `ST-LINK` or other debug interface?

prusnak, 2020-05-04:
Ok, the original issue is fixed. For the other problem, please open a new issue.

closed 2020-05-04 by prusnak

xlab, 2021-04-13:
Hey @Dimfred have you resolved your issue when Trezor T stucks in the starting up view? I've built firmware and getting the same problem as well. I tried to search other issues that would be a proper place to discuss this but found only this one lol.

Dimfred, 2021-04-13:
Hey @xlab, I think I solved it, but I can't recall how^^ It was just too long ago.

taherfattahi, 2023-05-06:
Hi @Dimfred, I've built firmware with legacy folder and now How can I sign my custom firmware?
I have trezor one and I can program bootloader with st-link but I can't install my custom firmware via trezor suite