trezor/trezor-firmware #981: Mark all security-related usage of bytes

tsusanka, 2020-05-06:
Let's create a new type "secbytes" to denote sensitive information.

prusnak, 2020-05-06:
This secbytes type should be immutable, not sure how that goes with MyPy and it's lax bytes/bytearray handling.

Here is the list of features (traits):
- immutable
- does have a `__add__` and `__radd__` method (for bytes and bytearray) returning secbytes
- can have automatic conversion from bytes/bytearray type to secbytes
- does not have a read method(!!!) - the only way how to convert secbytes to bytes/string should be via a method such as `publickey()`, see below

I pushed some of my random work into the [`prusnak/secbytes`](https://github.com/trezor/trezor-firmware/tree/prusnak/secbytes) branch. Some examples of typed functions:

```
def generate_secret() -> secbytes:
def publickey(secret_key: secbytes) -> bytes:
```

prusnak, 2020-05-08:
I just pushed a preliminary implementation of `secbytes` class to the [`secbytes`](https://github.com/trezor/trezor-firmware/tree/secbytes) branch in d912e2a2819a8bba61533b7036ee582cedd243c2.

It has all four properties (traits) I listed above, but nothing else (by design).

That's how one uses it:

```python
>>> from trezorcrypto import secbytes
>>> a = secbytes(b"ahoj")
>>> a
secbytes{id=1, len=4, data="61686f6a"}
>>> b = secbytes(b"babi")
>>> b
secbytes{id=2, len=4, data="62616269"}
>>> a + b
secbytes{id=3, len=8, data="61686f6a61686f6a"}
>>> c = a
>>> c
secbytes{id=1, len=4, data="61686f6a"}
>>> c += b
>>> c
secbytes{id=4, len=8, data="61686f6a61686f6a"}
>>> d = secbytes()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: missing argument
```

(print method will not work in the release build of course, I only added it here to show what's happening under the hood)

There is no way how to access the underlying data from the application code in Python. Application code can only create new instances of the class providing an argument with a buffer interface (`str`, `bytes`, `bytearray`) - the contents are copied into the `secbytes` internal data. Application code can also perform concatenation using the `+` operator.

However, C functions from `modtrezorcrypto` can detect whether the type is `&secbytes_type` and access the internal `secbytes` data using the following code in that case:

```python
mp_obj_t foo(mp_obj_t arg) {
    mp_obj_type_t *arg_type = mp_obj_get_type(arg);
    if (arg_type != &secbytes_type) {
        mp_obj_secbytes_t *sec = MP_OBJ_TO_PTR(arg);
        # access the data via `sec->ptr` and `sec->len`
    } else {
        # not secbytes, use the buffer interface (`mp_get_buffer_raise`) to read the data
    }
}

prusnak, 2020-05-08:
The next step would be to create a context manager class and modifying the `secbytes` constructor so it fails if it is called outside of a context manager. This CM collects all instances of `secbytes` created inside its block and removes them once the block is exited. Heck, the CM can even encrypt the data inside of the `secbytes` structure, so even the structure itself can't know what's inside. I have some ideas on how to do that.

matejcik, 2020-05-11:
we could start doing this purely on the typing side. something like `secbytes.pyi`
```python
class secbytes:
	def __init__(self, from: bytes) -> None:
		...

	def __add__(self, right: bytes) -> "secbytes":
		...

	def __radd__(self, left: bytes) -> "secbytes":
		...
```
(those are literal `...` btw)

then modify C extmod signatures to return secbytes, e.g.:
```python
from secbytes import secbytes

class HDNode:
	# ...
	def private_key(self) -> secbytes:
		...
```
then mypy should raise errors in all places that use result of `node.private_key` in a way that is not compatible with the very limited `secbytes` interface

matejcik, 2020-05-11:
implementation-wise, re contextmanager, there are two options I see:

(a) `secbytes` implementation itself is a context manager, and you need to call `__enter__()` in order to start using the value; before `__enter__`, all usages throw exceptions.
That probably won't work though, because the secbytes already hold the value, and if nobody ever calls __enter__, that value remains in memory.

(b) "secbytes factory":
```
with SecureContext() as s:
    b = s.secbytes(b"hello")
```
then all C extmod functions returning secbytes would need to accept a `SecureContext` instance as an argument.

There is no way to access the "current" context manager, and if we set a global implicit one in the background, we'd have a problem when secure contexts are nested.
I mean ... we _could_ implement a stack in the global implicit context manager, but that just seems weird. Creating and passing the instance seems like the way to go.

prusnak, 2020-05-11:
```
with SecureContext() as s:
    b = s.secbytes(b"hello")
```

Yes! That's what I meant.

matejcik, 2020-05-11:
btw this might pose a problem: http://docs.micropython.org/en/v1.9.3/pyboard/genrst/core_language.html#generator

Async functions are generators, and we have a lot of them.
IIUC this only applies to generators that are stopped, not generators interrupted by an exception, in which case we should be OK. But we will need to ascertain that.

prusnak, 2020-05-13:
In 703255a0f034bcaa83226fa1005ba73fbf38a65c I pushed the `SecureContext` manager. Demo:

```
>>> from trezorcrypto import SecureContext
>>> with SecureContext() as sc:
...     a = sc.new(b"AAAA")
...     b = sc.new(b"BBBB")
...     c = sc.concat(a, b)
...     print(a)
...     print(b)
...     print(c)
secbytes{valid=2863311530, len=4, data="41414141"}
secbytes{valid=2863311530, len=4, data="42424242"}
secbytes{valid=2863311530, len=8, data="4141414142424242"}
>>> print(a)
secbytes{valid=0, len=0, data=""}
>>> print(b)
secbytes{valid=0, len=0, data=""}
>>> print(c)
secbytes{valid=0, len=0, data=""}
```