trezor/trezor-firmware #982: Remove older versions of Zcash?

tsusanka, 2020-05-06:
In Zcash we currently support both `tx.version` 3 and 4. 3 should be Overwinter, 4 Sapling. Does it make sense to support both or is one of them obsolete?

We could simplify a few things if we could remove one of the versions. See [sign_tx/zcash.py](https://github.com/trezor/trezor-firmware/blob/3de565c33c3647d8ae776af2c02bc39bd602604d/core/src/apps/wallet/sign_tx/zcash.py) for some context.

@ca333 I believe you did some zcash contributions to our firmware any chance you could shred some light into this?

---

Also should we call the class `Zcash` instead of `Overwinter`?

---

Related to #617

prusnak, 2020-05-06:
AFAIK we can nuke `tx.version==3` and we can even put hard check whether `tx.version==4` for Zcash. I think Komodo is also on version 4 already.

tsusanka, 2020-07-03:
Since we are now streaming prev tx we need to support also legacy transactions. So it seems we won't be able to do this anytime soon? Closing? cc @matejcik

matejcik, 2020-07-07:
nope, this issue is about removing support for *signing* v3 transactions.
we already dropped v1 and v2 before the refactor, I think?
to close this, we need to find out if v3 transactions are valid on currently existing Zcash-like networks.
we *think* not, but it would be super nice to have some confirmation from actual Zcash / Komodo / Koto devs

tsusanka, 2020-07-07:
@matejcik I have confirmed this via email from Zcash. The v4 transaction format was required following the Sapling network upgrade. I do not know about Komodo/Koto though.

matejcik, 2020-07-14:
From a random sampling, both Komodo and Koto chains are using v4 since last year. That at minimum means that v4 transactions _are valid_ on the networks, and so we can safely drop v3 support, IMHO.

So let's go ahead and do this.

Heads-up for @ca333 for Komodo and @wo01 for Koto; please let us know if this would pose any problems for you.

wo01, 2020-07-14:
No problem.

PR #1146 (2020-07-30): zcash: drop support for signing v3 transactions
Fixes #982.

Also renames the `Overwintered` class to `Zcashlike`.
closed 2020-07-30 by tsusanka