trezor/trezor-firmware #984: Add bitshares and other graphene based chains

Dimfred, 2020-05-09:
Hello,
@xeroc  and me would like to contribute and try to find a way how to add graphene based chains to trezor.

**Is your feature request related to a problem? Please describe.**

From reading #219 I understood that you currently don't want to add graphene based chains, due to the huge difference in serialization formats. Could you please elaborate why? Is the problem for you that for each coin an app has to be added to the firmware, with all the necessary serializations?

**Describe the solution you'd like**

Would you consider to add graphene based chains if there was some graphene base layer, which could serialize transactions based on a predefined serialization table?

Consider the following example with two transfer operations from different chains:

    graphene_chain_1: transfer
    
    definition chain_1:
    {
        "name": "transfer",
        "serialization": [
            [
                "variable": "from", // name of the variable (protobuf_field will be created from that)
                "protob_type": "string",
                "serialization_type": "GrapheneObjectId", // the serialization has to be defined in a graphene app
            ],
            [
               "variable": "to",
               // rest same as above
            ],
            [
               "variable": "amount",
               "protob_type": "GrapheneAsset",
               "serialization_type": "GrapheneAsset"
            ]
        ]
    }

    definition chain_2:
    {
        "name": "Chain2Asset",
        "serialization": [
            [
               "variable": "amount",
               "protob_type": "uint32", // normal asset would be uint64
               "serialization_type": "Uint32"
           ],
           [
               "variable": "asset",
               "protob_type": "string",
               "serialization_type": "GrapheneObjectId"   
           ]
      ]
    }
    {
       "name": "transfer",
       .... same as chain_1 but using an its Chain2Asset definition in the amount field
    }

Signature / Operation preview could be performed in the same way. Basic operations could be easy defined this way. 

The protobuf_messages could also be created automatically this way. This sort of still doesn't solve the problem that for each operation a serialization format has to be defined.

If you would at least consider it I would do more reserach on the how much the grpahene chains out there differ.

@Roznovjak

prusnak, 2020-05-09:
As stated in the issue you are referring to - we are not interesting in adding coins that are not in TOP 30 on CMC.

closed 2020-05-09 by prusnak

xeroc, 2020-05-09:
Trying to clear this up in simple words:
What we would do is this:

* Allow the trezor to not only serialize a transaction, but also *de-serialize* it
* Send properly serialized transactions over USBwire
* let the trezor de-serialize the tx according to a *use-this-schema* message
* present data to the user
* re-serialize according to the schema, sign and send signature via USBwire.

**Advantages**:
* Users can still see the payload of transactions and operations
* We could support *all* features of any graphene chain by just uploading a JSON schema (as part of the signed firmware of course)
* Trezor can validate and confirm serialization of txs

@prusnak Thanks for your remark, I understand that. However, the intend of this issue is to bring forward a format that allows to support all graphene chains including STEEM, HIVE, EOS, BitShares etc ... not just a single one. We'd like to figure out a way to bring unreleased chains to Trezor too. So this ticket is not specific to BitShares, but to any Graphene-based blockchain.

xeroc, 2020-05-09:
Also, not sure you remember me from my profile picture: We've met during the Trezor Hackathon where I implemented STEEM support on Trezor One - fun times :D

prusnak, 2020-05-09:
I do remember you, yeah, it was nice event!

[EOS is supported](https://github.com/trezor/trezor-firmware/tree/master/core/src/apps/eos). Other coins you mention are not in the TOP 30.

I can't stop you from refactoring the EOS app to be more universal, but honestly, our priority is Bitcoin, so I can't promise we'll have time to review and merge.

xeroc, 2020-05-09:
> I can't stop you from refactoring the EOS app to be more universal, but honestly, our priority is Bitcoin, so I can't promise we'll have time to review and merge.

Fully, understand that.

I just wanted to see there is *potential* interest and that we are not running against a general *no-go* rule or similar. There needs to be good reasons to put resources into development **and** reviewing.

Let's rephrase this ticket a little:

Is there a general misconception in making Trezor compatible with Graphene by using the serialized transaction format on the USB wire and let Trezor do **de-serialization** together with serialization? That would allow us to make thing quite simple. All I need to know is if that approach would be an o.k. approach for you or if there is some sort of general *no-go* rule that would prevent this particular approach.

prusnak, 2020-05-09:
> by using the serialized transaction format on the USB wire and let Trezor do de-serialization together with serialization? 

This would include a lot of unnecessary logic into the Trezor code which would need to be maintained there, so no. We try to avoid de-serialization in Trezor code at all costs.

xeroc, 2020-05-09:
> We try to avoid de-serialization in Trezor code at all costs.

That's unfortunate