trezor/trezor-firmware #987: [Epic] Firmware translations (localisation)

tsusanka, 2020-05-13:
We would like to translate both T1 and TT firmwares to help onboard non-English users. We will first tackle this in Core, if _porting_ goes well, we will achieve this in T1 using the Core codebase.

What is most likely obvious but seems better to write it explicitly is that a incorrect translation could lead to a security issue where the user is confirming a different action than intended. We need to be careful.

## List of sub-issues:
- To Be Added

mmilata, 2020-11-02:
Note: depending on the chosen languages we might have to add additional characters to our fonts.

Hannsek, 2023-02-09:
We are moving to the milestone https://github.com/trezor/trezor-firmware/milestone/40

closed 2023-03-28 by Hannsek