trezor/trezor-firmware #988: Store signed data blob in storage

tsusanka, 2020-05-13:
During discussions on #987 @prusnak came up with an idea to use one whole memory block in Trezor's [storage](https://docs.trezor.io/trezor-firmware/hardware/model-t/memory.html) for signed blob of data received from the Host. The blob will contain mainly the translation strings for the chosen language, however its usage might be broader (e.g. coininfo etc.). We currently assume it would have 128KB.

This blob can be replaced only as a whole. We might split the blob into parts, provide hashes for each part and update only the parts that has changed. However, since uploading 128KB should take about 1s (fair guesstimate) this should be quite seamless for the user. We will consider this optimization later then, if necessary.

This issue covers:
- [ ] Figure out the blob signing process.
  - Are we going to use another key than what is used to sign the firmware? Probably yes.
  - Should this be auto-signed on Gitlab? Probably no.
  - We might come up with a similar mechanism as firmware has but lower the signature threshold. In another words, only one of @prusnak / @slush0 is required to generate a valid signature.
- [ ] Implement exchange mechanism where the Host sends the blob, Trezor verifies the signature and stores it in its storage.
- [ ] Research whether this is applicable for T1 as well.
- [ ] Research what the is the appropriate blob size. Is 128K the correct size?

prusnak, 2020-05-13:
I came up with the following blob structure: https://github.com/trezor/trezor-firmware/blob/master/docs/storage/addon-block.md 

Let's discuss this further whether this meets our needs.

prusnak, 2020-05-29:
Push the design of the i18n-block + modtrezori18n prototype to the[`i18n-block`](https://github.com/trezor/trezor-firmware/tree/i18n-block) branch.

closed 2020-05-29 by prusnak