trezor/trezor-firmware #989: Replace all English strings with identifiers and read them from the data blob

tsusanka, 2020-05-13:
We need to replace all UI texts with identifiers. Firmware will then have no language and we will load the correct strings from the data blob (as defined in #988).

This issue covers:
- Replacing all English text with identifiers.
- Use the sentences stored in the data blob fetched using the identifier.
- This should be a completely internal change, user should not notice any changes.

---

Needs #988. Part of #987.

matejcik, 2020-05-13:
notes:

* For development convenience, this should be automated as much as possible.
  Also consider that identifiers need to be `import`ed from somewhere, otherwise we don't have a valid Python.

Implementation-wise, what I would like most is using gettext-like operations:
```python
from gettext import gettext as _

def something():
	display.show(_("This string will be translated"))
```
We can then automate the build process so that either `"This string will be translated"` converts to a key in the hashtable, or the whole call is replaced.
This can be easily and reliably done through [libcst](https://github.com/Instagram/LibCST) -- which can also extract the list of strings for us. (but gettext can do that too)

Maybe it's a good idea to replace the `BITCOIN_ONLY` sed with libcst as well?

* What about plurals?

* What about format strings? We should unify format string style across the codebase, at least for translated strings.

matejcik, 2020-05-14:
One drawback of using gettext-like strings is that when the string changes, the lookup key changes as well. I'm not sure what to do about that just yet.

We could convert the codebase to gettext-like and then do one pass to convert something like `_("This string will be translated")` to `_("TR-this-string-will-be")`, with the latter being a fixed key.

Or we could keep a lookup table from English-in-codebase strings to some kind of numeric identifiers, and have the build system catch us when new strings are added. We might want that *anyway* to simplify lookup in the translation blob.

tsusanka, 2020-05-26:
Let's do this with #991 it is almost a duplicate. Sorry for the confusion.

closed 2020-05-26 by tsusanka